bbcmicrobit/micropython #218: Weird issue with music wait=False

gbaman, 2016-02-24:
Was fiddling around with doing piano stuff, where the note doesn't stop till something else stops it (holding down the key on the piano then let go).
I had this code below.
```python
from microbit import *
import music
print("Starting")

sleep(1000)

music.play('c4:4', wait=False)

while True:
    print("Test")
    sleep(1000)
```
When I ran it on the Microbit using Mu, got a bit of a surprising result... Is only an issue when I set ```wait=False```, which is required to do my idea and use ```.stop()``` later on.
[image]

dpgeorge, 2016-02-24:
Yep, that is weird!  Likely it's something to do with trying to convert an object to a string to play the note, but instead of a string there is a boolean.  The strange part is, where does the boolean come from?  I'll look into it.

dpgeorge, 2016-02-25:
Work around is to put the single note in a list:
```
music.play(['c4:4'], wait=False)
```

ntoll, 2016-02-25:
Fixed by #219

closed 2016-02-25 by ntoll