bbcmicrobit/micropython #222: We need a test suite

dpgeorge, 2016-02-26:
It goes without saying that we need a bunch of tests that make sure the microbit API remains stable and doesn't regress, especially with things like #221.

MicroPython upstream has a big test suit for Python features, so we don't need to duplicate that (or you can just run that test suite on the microbit, it mostly works).  We need tests for things like display rendering, accelerometer, gestures and music.

deshipu, 2016-02-26:
I wonder how you even test a hardware project such as this? Wire it up to a second microcontroller and verify that the outputs are what you expect?

JoeGlancy, 2016-02-26:
Is there an M0 emulator which might exist? a A universal test suite that
everyone could use might be a but complicated if it required an external
microcontroller, unless it was something commonplace like an Arduino which
a lot of people have.

dpgeorge, 2016-02-26:
There are various things you can do to test hardware: at the least you just want to verify that the functions can be called as specified by the API docs.  Then you can test some things like accelerometer and buttons by prompting the user.  You can also connect pins to other pins and test GPIO that way.  Connecting to another microcontroller is the only way to fully test everything, but is probably overkill.

JoeGlancy, 2016-02-26:
You could have a `selftest` module even, which does

dpgeorge, 2016-04-09:
A simple test suite (just one script at the moment) has been added.

closed 2016-04-09 by dpgeorge