bbcmicrobit/micropython #228: I2C speed

carlosperate, 2016-03-07:
There is no mention anywhere about the default clock speed of the I2C interface, and the current interface does not allow to change this. As both the accelerometer and compass are normal and fast mode compatible, could it be possible to add that option to the I2C module? Does the DAL currently support that?

dpgeorge, 2016-03-08:
The default I2C clock speed is 100kHz.  The DAL does not support changing the frequency, but the underlying mbed interface does.  We can expose this, eg i2c.init(freq=250000).

carlosperate, 2016-03-08:
Being able to set the frequency would be quite useful. I'm not quite sure
how the mbed interface relates to micropython (I assume both share the same
DAL), is it currently possible to implement it?

dpgeorge, 2016-03-09:
> is it currently possible to implement it?

Yes, rather easy to do.

dpgeorge, 2018-03-07:
Commit e88e3b2f6c579eab0eafda02242092bff51f6934 added an option in i2c.init() to set the frequency.

closed 2018-03-07 by dpgeorge