bbcmicrobit/micropython #230: OpenSSL dependancy needs adding to doc

WayneKeenan, 2016-03-10:
On Ubuntu Trusty I received this error whilst installing Yotta

`build/temp.linux-x86_64-3.4/_openssl.c:431:25: fatal error: openssl/aes.h: No such file or directory`

It was necessary to run the following before installing Yotta 

`sudo apt-get install libssl-dev`

JoeGlancy, 2016-03-10:
Strange, I never got an error like that, and my system was completely clean
(VM).

dpgeorge, 2016-03-10:
It could be that yotta was recently upgraded and now relies on libssl.  Can anyone else confirm this?

komeP, 2016-06-02:
yotta depends on cryptography now, which depends on the OpenSSL headers.

carlosperate, 2016-06-02:
Last time I had to install yotta over a month or two ago it already needed libssl-dev.

carlosperate, 2017-09-19:
@dpgeorge  or @ntoll, this is already in the README, so this issue can be closed.

closed 2017-09-19 by ntoll