bbcmicrobit/micropython #237: Beeps and Boops - Piezo's need a square wave

descartes, 2016-03-13:
That sort of buzzer would need to be energised by a square wave around 2kHz - follow the link you've given for SparkFun to confirm.

You can either use a low current buzzer - I'll have a look for one. Or use the PWM output on a pin to drive the piezo. Or use suitable sleep values in a loop to switch the pin on & off at some reasonable rate (see below).

For this easy to understand example, the buzzer would be the best route.

The piezo will actually run at a wide range of frequencies but the element is tuned for a particular frequency - I use 2kHz piezo's and can get a reasonable tune out of them (middle C is 262Hz), but the volume goes right up when you hit the tuned frequency.

dpgeorge, 2016-03-13:
I think this "Beeps and Boops" tutorial is talking about a buzzer that just requires power to make a (fixed freq) tone.  You are right that the image of the piezo refers to something that is not a "buzzer".  @ntoll do you know which piece of hardware you used for this tutorial?

ntoll, 2016-03-13:
Greeting from Bratislava! OK... this is my mistake, and many apologies. I Googled for Piezo buzzer and used the (liberally licensed) image found as a result. It certainly looks like the hardware I used for this tutorial but I'll need to check the details when I return home. Mine is certainly a fixed pitch buzzer.

cefn, 2016-05-28:
I was just about to file this as an issue and found this thread. To clarify, the difference is whether the component is a Piezo transducer/speaker/passive buzzer (voltage input translates to movement) or an active buzzer (voltage input drives an oscillating circuit, which generates a voltage which translates to movement).

This example is especially problematic since housed Piezo transducers are a common way of wiring up a circuit suitable for e.g. http://microbit-micropython.readthedocs.io/en/latest/tutorials/music.html as we do with our Alarm Clock... http://start.shrimping.it/project/alarmclock/build.html#step12 despite the misleading warning in the Music tutorial...

```Do not attempt this with a Piezo speaker - such speakers are only able to play a single tone.```

...which should probably also be removed as using a Piezo transducer is fine and probably recommended as they are cheap, not too demanding of current, and ideal for classroom wiring.

If you wanted to reference a Piezo transducer in the music example, we reference our suppliers for every kit component e.g. links from http://start.shrimping.it/kit/alarmclock.html From experience, there seems to be very little consistency in naming among cheap suppliers. In the past I've bought elements which I expected to be transducers and they turned out to be active buzzers

So if you want to retain this simple Input/Output example, describing it as an active buzzer could help. However, I would be tempted to drop the example altogether, as it has very little to do with Bleeps and Bloops, and a Piezo Active Buzzer could equally be a series resistor and LED given the logic of the circuit and code.

If the example was dropped, the Music tutorial could link to suitable Piezo transducers instead without confusion, and the misleading caveat from that tutorial could be removed too.

ntoll, 2016-05-28:
Hi folks, sorry this fell off my radar. I'll correct this in an appropriate way so there's no confusion between a fixed pitch buzzer and speaker. Expect a PR soon.

ntoll, 2016-05-28:
See #285 :-)

dpgeorge, 2016-05-29:
> I would be tempted to drop the example altogether, as it has very little to do with Bleeps and Bloops, and a Piezo Active Buzzer could equally be a series resistor and LED given the logic of the circuit and code.

If you use an LED in the example then you need to talk about resistors and resistor values.  The piezo buzzer is simpler albeit probably a less common component.

ntoll, 2016-06-01:
Merged #285 to correct the misunderstanding.

closed 2016-06-01 by ntoll