bbcmicrobit/micropython #239: Yotta Build Issue (File Missing)

vlee489, 2016-03-13:
There is a problem when building Micropython for the Microbit. As well some minor errors came for a ride.

Platform: Windows 10 64x

    c:\Users\Vincent\Documents\MicroBit\micropython - build>yt target bbc-microbit-classic-gcc-nosd
    info: get versions for bbc-microbit-classic-gcc-nosd
    info: download bbc-microbit-classic-gcc-nosd@0.1.4 from the public module registry
    info: get versions for mbed-gcc
    info: download mbed-gcc@0.1.3 from the public module registry

    c:\Users\Vincent\Documents\MicroBit\micropython - build>yt up
    info: get versions for bbc-microbit-classic-gcc-nosd
    info: get versions for mbed-gcc
    info: get versions for mbed-classic
    info: download mbed-classic@0.0.4 from the public module registry
    info: get versions for microbit-dal
    info: download microbit-dal@v1.4.15 from GitHub lancaster-university/microbit-dal
    info: get versions for ble
    info: download ble@v2.5.0+mb2 from GitHub lancaster-university/BLE_API
    info: get versions for ble-nrf51822
    info: download ble-nrf51822@v2.5.0+mb4 from GitHub lancaster-university/nRF51822
    info: get versions for nrf51-sdk
    info: download nrf51-sdk@v2.2.0+mb2 from GitHub lancaster-university/nrf51-sdk

    c:\Users\Vincent\Documents\MicroBit\micropython - build>yt build
    info: generate for target: bbc-microbit-classic-gcc-nosd 0.1.4 at c:\Users\Vincent\Documents\MicroBit\micropython - build\yotta_targets\bbc-microbit-classic-gcc-nosd
    GCC version is: 5.2.1
    suppressing warnings from ble-nrf51822
    suppressing warnings from nrf51-sdk
    suppressing ALL warnings from mbed-classic, ble, ble-nrf51822 & nrf51-sdk
    -- Configuring done
    -- Generating done
    -- Build files have been written to: C:/Users/Vincent/Documents/MicroBit/micropython - build/build/bbc-microbit-classic-gcc-nosd
    [141/279] Building CXX object ym/microbit-dal/source/CMakeFiles/microbit-dal.dir/ble-services/MicroBitBLEManager.cpp.o
    C:/Users/Vincent/Documents/MicroBit/micropython - build/yotta_modules/microbit-dal/source/ble-services/MicroBitBLEManager.cpp:170:71: warning: unused parameter 'serialNumber' [-Wunused-parameter]
     void MicroBitBLEManager::init(ManagedString deviceName, ManagedString serialNumber, bool enableBonding)
                                                                       ^
    [145/279] Building CXX object ym/microbit-dal/source/CMakeFiles/microbit-dal.dir/MicroBitMessageBus.cpp.o
    C:/Users/Vincent/Documents/MicroBit/micropython - build/yotta_modules/microbit-dal/source/MicroBitMessageBus.cpp: In member function 'int MicroBitMessageBus::deleteMarkedListeners()':
    C:/Users/Vincent/Documents/MicroBit/micropython - build/yotta_modules/microbit-dal/source/MicroBitMessageBus.cpp:187:61: warning: suggest parentheses around operand of '!' or change '&' to '&&' or '!' to '~' [-Wparentheses]
         if (l->flags & MESSAGE_BUS_LISTENER_DELETING && !l->flags & MESSAGE_BUS_LISTENER_BUSY)
                                                                 ^
    [162/279] Building C object source/CMakeFiles/microbit-micropython.dir/C_/Users/Vincent/Documents/MicroBit/micropython_-_build/source/lib/neopixel.c.o
    In file included from c:/Users/Vincent/Documents/MicroBit/micropython - build/inc/lib/neopixel.h:34:0,
                     from C:/Users/Vincent/Documents/MicroBit/micropython - build/source/lib/neopixel.c:80:
    c:/Users/Vincent/Documents/MicroBit/micropython - build/yotta_modules/nrf51-sdk/source/nordic_sdk/components/drivers_nrf/delay/nrf_delay.h:194:1: warning: 'inline' is not at beginning of declaration [-Wold-style-declaration]
     static void __INLINE nrf_delay_us(uint32_t volatile number_of_us) __attribute__((always_inline));
     ^
    c:/Users/Vincent/Documents/MicroBit/micropython - build/yotta_modules/nrf51-sdk/source/nordic_sdk/components/drivers_nrf/delay/nrf_delay.h:195:1: warning: 'inline' is not at beginning of declaration [-Wold-style-declaration]
     static void __INLINE nrf_delay_us(uint32_t volatile number_of_us)
     ^
    [279/279] Linking CXX executable source/microbit-micropython
    FAILED: cmd.exe /C "cd . && C:\PROGRA~2\GNUTOO~1\5F17D~1.220\bin\AR10B2~1.EXE   -fno-exceptions -fno-unwind-tables -Wl,--gc-sections -Wl,--sort-common -Wl,--sort-section=alignment -Wl,-wrap,main -mcpu=cortex-m0 -mthumb -TC:/Users/Vincent/Documents/MicroBit/micropython - build/yotta_targets/bbc-microbit-classic-gcc-nosd/CMake/../ld/NRF51822.ld -Wl,-Map,source/microbit-micropython.map -Wl,--start-group @CMakeFiles/microbit-micropython.rsp   -lstdc++ -lsupc++ -lm -lc -lgcc -lstdc++ -lsupc++ -lm -lc -lgcc -Wl,--end-group  --specs=nano.specs -o source/microbit-micropython && cmd.exe /C "cd /D "C:/Users/Vincent/Documents/MicroBit/micropython - build/build/bbc-microbit-classic-gcc-nosd/source" && arm-none-eabi-objcopy -O ihex microbit-micropython microbit-micropython.hex && srec_cat microbit-micropython.hex -intel -o microbit-micropython.hex -intel --line-length=44 && cd /D "C:/Users/Vincent/Documents/MicroBit/micropython - build/build/bbc-microbit-classic-gcc-nosd/source" && "C:/Program Files (x86)/GNU Tools ARM Embedded/5.2 2015q4/bin/arm-none-eabi-objcopy.exe" -O binary microbit-micropython microbit-micropython.bin""
    AR10B2~1.EXE: error: build/yotta_targets/bbc-microbit-classic-gcc-nosd/CMake/../ld/NRF51822.ld: No such file or directory
    ninja: build stopped: subcommand failed.
    error: command ['ninja'] failed

dpgeorge, 2016-03-13:
This is probably an issue for yotta, not here.  I don't know if Windows is supported.

On Linux, the file you need is located in `./yotta_targets/bbc-microbit-classic-gcc-nosd/ld/NRF51822.ld`

dpgeorge, 2016-04-09:
I assume this was fixed upstream.

closed 2016-04-09 by dpgeorge