bbcmicrobit/micropython #245: Create radio module based on DAL implementation

ntoll, 2016-03-26:
What the title says. This should be a separate module so the radio feature is not automatically available to kids (they have to explicitly import it).

damianburrin, 2016-05-22:
What us the concern @ntoll ?

ntoll, 2016-05-22:
The BBC have a concern surrounding child protection. I believe this shouldn't stop us implementing this feature since I believe it spurious.h

damianburrin, 2016-05-22:
>The BBC have a concern surrounding child protection
Oh ok.  Seems a bit over the top to me. You'd need to have a program instalked to recieve any form of message and then from fairly close. 
I hope to be able to use this feature in python rather than C.  Ive got a neat idea of a classroom 'Exit Pass'

ntoll, 2016-05-23:
Exactly... OTT.

It's coming.

markshannon, 2016-09-19:
We have a radio module now.

closed 2016-09-19 by markshannon