bbcmicrobit/micropython #246: Create "filesystem" (local storage) module based on DAL implementation

ntoll, 2016-03-26:
What the title says.

markshannon, 2016-03-30:
The persistent module referenced above was designed to store audio samples, so doesn't have a file-like API.
I've just had a look at the DAL PR: https://github.com/lancaster-university/microbit-dal/pull/110 and it seems to be rather inflexible in terms of file size and number, and rather inefficient as well. It may also cause problems with the hardware by repeatedly writing to the same areas of the flash.

As to the API, I would suggest:
```python
open(name, mode='r') #returns a file descriptor
remove(name) # Removes a file
fd.read(len=None) # Read len bytes (or the whole file if len is None)
fd.write(buf) # Writes the bytes in buf to the file.
fd.close() # Writes any necessary metadata to the file-system.
```
I have a prototype implementation built on top of https://github.com/bbcmicrobit/micropython/pull/249 that organises 20kb of flash into 80 256-byte chunks, with 254 usable bytes in each, Files can be formed of any integral number of chunks. The design aims to keep erasure and writing to a minimum and avoid any hotspots which might lead to hardware failure.

It might be nice to automatically execute any file called `main.py` on reset.

larryhastings, 2016-03-31:
I suspect you want it more granular than that--perhaps 160 128-byte chunks, 124 usable bytes in each?  There's a certain amount of overhead for packages (empty __init__.py files).  Also, if this is used to store precompiled bytecode files those will often be quite small.

Would it be possible to drop the "chunk" approach completely and simply pack the data as tightly as possible?  One-byte length, filename, two-byte length, file data?

markshannon, 2016-03-31:
The problem with packing the data is fragmentation.

Given the per-chunk overhead is only 2 bytes, then reducing the chunk size to 128 would make sense. There is an upper limit of 253 chunks (1 byte index - 3 special values) per "file system", so 64 byte chunks would be too small.
Another thing to consider is that the flash must be erased in 1k pages, although I doubt it make much difference whether the chunks are 128 or 256 bytes.

larryhastings, 2016-03-31:
ISTM there are two use cases for a "filesystem" on the micro:bit:

1. Storing Python source files, data tables, and other read-only data.
2. Storing dynamically-generated data (high scores, accelerometer telemetry).

I had only the first use case in mind, I gather you were thinking of both.

I suggest that, ideally, these two very different use cases might benefit from different implementations.  For 1, packing the data will let us maximize storage.  For 2, your chunked approach (and a simple wear-leveling algorithm) would be better.

Might it make sense to have both?  The actual look-up-the-data portion of the filesystem code will be small, so having two wouldn't be a big deal.  Obviously they'd behave like a single filesystem, and writing a file to the read-write area would occlude any read-only version.

whaleygeek, 2016-03-31:
Also note that there is a significant difference (in implementation,
performance and wear levelling) comparing append-only writing and random
writing.

Mostly, data logging will be append only. Reading and writing config
settings or min/max states would be random. Reading stored scripts would be
read only and mostly sequential from start to end.

For things like min/max state changing, a simple transactional file system
(i.e. with delete markers - sort of an upside down stack) gives great wear
levelling at the cost of slightly increased first-read time (depending on
the read algorithm).

It's worth checking the data sheet for the NRF to see if the flash supports
multi-write (i.e. can you write to a location again without an erase) -
some persistent devices do support this, providing you are only ever
changing a '1' bit to a '0' bit, and you can implement some really simple
and tiny transactional files using this to your advantage (I have done on
many tiny micro controllers in the past). The cost is often a single full
file scan on first read, to find the newest record (so that you don't have
to persist start/end pointers in memory and, erm, burn out the cell for
those) and you can persist this in RAM so that the next read/write is fast.

closed 2016-05-17 by dpgeorge