bbcmicrobit/micropython #248: Display.Input ()

damianburrin, 2016-03-26:
Both the touch developer and the block editor have a light level feature that reads the exterior light level via the leds.

Could this be added to the MicroPython version?

markshannon, 2016-03-27:
Does it work?

damianburrin, 2016-03-28:
It does in the touch and blocks version but doesn't seem to be supported in the python tool set

markshannon, 2016-03-29:
Sorry, I wasn't clear there. 
Does it provide reliable readings of the external light level?

damianburrin, 2016-03-29:
Yes in the block and touch it gives a value from 0 to 255 ( so i guess theres an 8bit adc somewhere)  as you flash a light on the leds (un lit) thee value rises which can be read via a block or touch code.  ive done a cpl of mini programs using them one displays a 'graph' based on how much light is being recieved the other is a theramin that increases the pitch of a tone based on light.

Works well but would prefer to do it in python with the students.

Damian

markshannon, 2016-03-29:
OK, thanks.

bjost2s, 2016-11-05:
I am also looking for the light level function, is it already implemented?

ZanderBrown, 2017-02-06:
Any progress on this?
Obviously it wouldn't be especially accurate but would be good to have.

microbit-carlos, 2018-12-14:
`display.read_light_level()` method was added in https://github.com/bbcmicrobit/micropython/commit/02ef4d03bef9a72f60bb7f0f2ae1b90c75cd9432 and included in MicroPython v1.0.0.

closed 2018-12-14 by microbit-carlos