bbcmicrobit/micropython #256: exit repl?

carlosperate, 2016-04-16:
Maybe I am just missing some obvious thing, but how do we exit the REPL when it is being accessed form a terminal? (trying from linux, but really asking about all systems)

Ctrl+D resets the microbit and returns to the REPL
Ctrl+C stops a running program in micropython
Ctrl+Z does nothing
Ctrl+A takes you to the "raw REPL" but then the above still apply (ctrl D seems to throw a syntax error?)
sys.exit() is not implemented either

I haven't read all the documentation, but using the search box couldn't really find the answer either.

At the time of testing I was running `MicroPython v1.6-11-gfa67b52 on 2016-02-01`.

dpgeorge, 2016-04-16:
It depends what terminal access program you use.  For picocom, you must type ctrl-A ctrl-X.

As a side note, sys.exit() is implemented: import sys; sys.exit().  That will terminate a script.

carlosperate, 2016-04-16:
Thanks for the quick response Damien!

Right, so sys.exit() is implemented on the script, but not in the REPL?

I must admit that I tried ctrl-A and then ctrl-X on picocom, but did so with a long pause in the middle (basically waited fro the raw REPL to be activated), for some reason I assumed it was a command parsed by micropython rather than picocom itself.

Okay, so just to confirm, in all cases we need to close the connection from the terminal software, rather than something that can be done by the REPL itself?

dpgeorge, 2016-04-16:
> Right, so sys.exit() is implemented on the script, but not in the REPL?

It's implemented in the REPL as well (no difference to the script, just raises SystemExit exception).

> Okay, so just to confirm, in all cases we need to close the connection from the terminal software, rather than something that can be done by the REPL itself?

Yes, the REPL/script can in no way force your terminal to close/exit/terminate.

carlosperate, 2016-04-17:
I see, thanks for the help @dpgeorge !

closed 2016-04-17 by carlosperate