bbcmicrobit/micropython #257: Button get_presses/reset_presses needs reworking

dpgeorge, 2016-04-17:
The idea of the `button.get_presses()` and `button.reset_presses()` methods is that you can get the number of queued presses and react on them (eg incr a counter).  But actually it's still possible to miss a press with these methods, namely:
```
i = button.get_presses()
# button is pressed here
button.reset_presses()
# i doesn't take into account the most recent press
```
To fix this requires a `button.get_and_reset_presses()` method.

I think it's bad if we let/encourage incorrect code like the above.  Instead we should provide proper atomic functions like `get_and_reset_presses` so people get used to the complexities of doing such things.

Gestures are also affected by the same thing.

markshannon, 2016-04-17:
The  `was_pressed` pressed method is the boolean version of `get_presses` and it resets itself. Perhaps `get_presses()` should do likewise and we can drop `reset_presses()`. Running totals can easily be obtained with `cnt += button.get_presses()`.
Will need the docs to updated, of course.

dpgeorge, 2016-04-17:
> Perhaps get_presses() should do likewise and we can drop reset_presses(). Running totals can easily be obtained with cnt += button.get_presses().

Yes, this seems like the most elegant way to do it.  Dropping the `reset_presses()` method makes the API simpler to use.  And I don't really see any need to have it a separate method.  You'd probably always call reset after get.

ntoll, 2016-04-17:
I like @markshannon's suggestion. Simple wins every time. ;-)

ntoll, 2016-04-19:
Fixed by #258

closed 2016-04-19 by ntoll

whaleygeek, 2016-04-19:
Thanks for breaking all my example programs :(

Real users are using this deployed code (even though you are not at 1.0).

ntoll, 2016-04-19:
@whaleygeek show me the code and I'll fix it.

markshannon, 2016-04-19:
@whaleygeek Sorry about that. Version 1.0 is coming very soon, so no more API changes thereafter.

Would you like to contribute some of your example programs? We could always do with more examples.

dpgeorge, 2016-04-19:
@whaleygeek unfortunately, that is the nature of progress :)

Where do you pull your source/binaries from?  Do you compile directly from this repo?

If needed we can add back a dummy, undocumented reset_presses() function.  That will un-break things.

markshannon, 2016-04-19:
I don't think adding an undocumented `reset_presses()` is a good idea.
Undocumented doesn't mean unsupported; it just means that we have a hole in the documentation.

dpgeorge, 2016-04-19:
> I don't think adding an undocumented reset_presses() is a good idea.
> Undocumented doesn't mean unsupported; it just means that we have a hole in the documentation.

It's the way to support legacy code/scripts.  Undocumented means that we don't want people to use it.  If you prefer, we can document it as deprecated :)

whaleygeek, 2016-04-19:
It's not a huge issue for me specifically. 

It's just that the micro:bit is out there in the wild, real people have written programs with python (many teachers, educators, and kids included), and if you deploy this fix, you will break those programs. I can fix mine later when I have time, and in the meantime if someone tries to download them and they break, they will have to work it out themselves.

But the fewer 'non backwards compatible' changes to the API you can make, the better, as there are several hundred thousand micro:bits out there already and it is growing daily.

JoeGlancy, 2016-04-19:
@dpgeorge having a dummy reset function would be useless as scripts could
still break/malfunction if they expect get_presses to not reset itself -
why not keep both, and have a new, separate get_presses_and_reset() or
similar, which combines both into one? Then @whaleygeek (and others) can
keep their scripts unmodified, or upgrade if need be.

dpgeorge, 2016-04-19:
> having a dummy reset function would be useless as scripts could still break/malfunction if they expect get_presses to not reset itself

I think most uses of these functions have get followed directly by reset.  Or a check if get returns zero, and if it returns non-zero then call reset.  Both of these cases will still work with the reset functionality moved to get, and reset made to do nothing.

If a script uses these functions in a different way than I describe above then probably it's already broken due to it being able to miss events.  It'll no be broken in a different way :)

> why not keep both, and have a new, separate get_presses_and_reset() or similar, which combines both into one?

Because we don't want to encourage use of get_presses/reset_presses as separate functions.

JoeGlancy, 2016-04-19:
Ah, gotcha. I assume it'll be removed eventually (reset_presses)?

whaleygeek, 2016-04-19:
I think my issue is more that there seems to be no official deprecation process to manage changes now the micro bit is massively deployed.

If I have an older version of Mu, and publish my old code on the web and in print form, it works. Then you change the API by taking one method out and putting another one in. So, all the while I (and others) are using the older version of Mu it works fine. But then someone new downloads the latest Mu, and the script does not work.

If I upgrade my script to use the new API, and download the new Mu, it solves the problem for me and all other people who have the new Mu. But for those that still have the old Mu, the instructions are wrong. So there are now two completely different incompatible universes.

It feels like just storing up problems, and what is worse, it may take an indeterminate amount of time (and different amounts of time) for people to notice this problem.

The other issue is that Mu does not (or did not?) allow you to save the .hex file to a location other than a microbit (unlike the web editors), so you don't have a cached copy anywhere of the actual binary that actually works, you have to reflash from Mu every time. So, you go running into a workshop on short notice with an upgraded Mu, you load up some code you haven't used for a while, and it falls over in front of a load of people.

My preference would be that this discussion turns into a separate 'how to manage the deprecation process', perhaps in a different issue number, as that is my point really.

I also have a preference for calling what is already deployed 1.0 before this change is made, and then managing the change via a proper deprecation process. You're only going to store up problems otherwise.

The micro:bit was launched on 22nd March and has been mass deploying ever since. I would recommend the version that was live on 22nd March is called 1.0, and all future changes are put against a 1.1 managed by a proper deprecation process. But that is only my opinion, as a user trying to deliver workshops with this thing and not much time to keep going back and fixing my resources and re-printing things. But there will soon be 1 million+ users of the thing.