bbcmicrobit/micropython #264: Allow to configure range (and rate?) of accelerometer

dpgeorge, 2016-05-03:
The accelerometer defaults to a 2g max range, but can do up to 8g.  The DAL provides a function to get/set the range (and sample rate) of the accelerometer and we should probably expose this in Python.  Eg:

```
accelerometer.set_range(8)
accelerometer.get_range()
accelerometer.set_rate(50) # 50Hz sample rate
accelerometer.get_rate()
```

catchpolej, 2016-10-06:
I am right in thinking that a reading of 1024 using default settings is equal to 'g'?

dpgeorge, 2016-10-06:
> I am right in thinking that a reading of 1024 using default settings is equal to 'g'?

The return value is is milli-g's, so 1000 is equal to 'g'.

microbit-mark, 2020-06-08:
This has been requested again recently in micro:bit support ticket: 37999

It looks like this PR is still open https://github.com/bbcmicrobit/micropython/pull/266 
is this something that we can implement? It seems a useful feature and would be comparable to our MakeCode editor.