bbcmicrobit/micropython #270: display.show parameters

gbaman, 2016-05-10:
Have by mistake stumbled across that if you don't specify the paramenters with display.show(), the result is an angry MicroPython:
``` Python
microbit.display.show("Hello", 400, True, False, False)

Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
TypeError: extra positional arguments given
```
Is this intentional? Of course could do ```microbit.display.show("Hello", 400, wait=True, loop=False, clear=False)```, but wasn't expecting to need to. I may be missing something, but seems a bit weird to force it that way.

dpgeorge, 2016-05-10:
The specification of the show method should be (the docs should be updated to match this):
```
microbit.display.show(iterable, delay, *, wait=True, loop=False, clear=False)
```
The `*` means that all following args must be specified as keywords.  The reason for doing it this way is to make it clear, when reading code, as to what all those booleans are used for.  If you don't have keywords then `True, False, False` could be anything, in particular how do I easily remember what the order is of these args?

markshannon, 2016-05-10:
@gbaman The `wait`, `loop` and `clear` parameters all have sensible defaults.

Instead of writing `microbit.display.show("Hello", 400, wait=True, loop=False, clear=False)`, you can simply write `microbit.display.show("Hello", 400)`.

gbaman, 2016-05-11:
@markshannon am aware of that, was more am working a project that will involve them all needing to be there, given will be changed.    
@dpgeorge given I didn't even know that syntax was a thing for documentation, I doubt a year 7 would understand it without extra explanation?
Meaning if one wants to set wait to False for example, they are just going to do
```microbit.display.show("Hello", 400, False)```
Given that is how most libraries work with optional parameters. Which will in turn throw a rather unhelpful error, which simply confuses them further.

ntoll, 2016-05-11:
So we should update out documentation to make the use of optional named parameters absolutely clear.

dpgeorge, 2016-05-11:
We should update the documentation.

The idea is that kids (and users in general) will take existing code (eg from examples in the documentation) and modify them to suit their needs.  Such examples will look like `show('hello', wait=False)` and the meaning of that code should be obvious to anyone, even those who don't know Python.  Also it should be easy to change the behaviour in this example.  But if the example looked like `show('hello', False)` then you'd need to read the docs to understand what the boolean argument is doing.

By forcing the use of keyword arguments we are (hopefully!) helping people to write clean and readable code, code that when shared is easy to understand and modify.

This is a good example of where we can benefit from the huge R&D investment done in normal Python.  They came up with keyword-only args for a good reason and we should follow suit.

JoeGlancy, 2016-05-11:
The docs are currently aimed more as an API/dev reference, and not
completely for children *yet* (as we don't have a stable release). At the
end though, it will need a large rework aimed at younger children with
examples for everything.

ntoll, 2016-05-11:
That's what the tutorials are for. Viz: https://microbit-micropython.readthedocs.io/en/latest/tutorials/introduction.html

dpgeorge, 2016-05-11:
See #271 for a docs update.

JoeGlancy, 2016-05-11:
Personally, I'd like to see an example of how to use each function at its
reference, instead of having to go through a tutorial and find what I
wanted (but most of the API is memorizable/easy to use anyway).

ntoll, 2016-05-11:
@JoeGlancy of course. We do accept PR's you know... ;-)

markshannon, 2016-11-07:
@c0d3st0rm Any specific suggestions as to how we can further improve the documentation, or can I close this issue?

microbit-carlos, 2018-03-13:
Closing as the documentation was updated and there was no more suggestions given.

closed 2018-03-13 by microbit-carlos