bbcmicrobit/micropython #274: Add display.disable() and display.enable()

markshannon, 2016-05-16:


carlosperate, 2016-05-16:
This could also be useful to enable touch support for those pins thanks to https://github.com/lancaster-university/microbit-dal/issues/90 
(would it have to get backported to 1.4.x? @finneyj offered to do that, but I am not quite sure what is the plan with updating the DAL version for micropython)

dpgeorge, 2016-05-16:
Might also be nice to be able to query the current state of the display, eg is_enabled().  But to avoid adding 3 very simple methods (disable, enable, is_enabled) how about a single method like display.active():
```
if display.active():
    print('it's active!')

display.active(True) # enable it
display.active(False) # disable it
```

> but I am not quite sure what is the plan with updating the DAL version for micropython

We need to decide on this.

markshannon, 2016-06-02:
After much discussion (at the PyCon sprints), we've decided that three functions are clearer than one.
`on()`, `off()`, and `is_active()` is the current favourite trio.
@dpgeorge Any compelling reason not to have three functions?

ntoll, 2016-06-02:
We should also use the same convention with the `radio` module since that's another thing that can be turned on and off.

dpgeorge, 2016-06-02:
> Any compelling reason not to have three functions?

Having only 1 method saves code size (don't need 3 interned strings, or 3 Python wrappers for the separate functions).  But if you had a long discussion and think that 3 are clearer then go for it!

But can you think of something even simpler for is_active(), eg is_on()?

ntoll, 2016-06-02:
+1 `is_on`

PR #289 (2016-06-03): Display on/off
At PyCon 2016, we are converging on using on()/off()/is_active() to set and query the display state. Activate/deactivate are relatively complex words, and using a single function seems a lot like a property.

When the display is off, state is preserved and animation is paused. When the display is turned back on, the state is restored and animation is resumed from the point where it was paused. This seems a reasonable behavior, and is perhaps simpler than discarding the state.

If the display state is modified when off, the effect of the modifications will be visible when the dis [... cut]
closed 2016-06-03 by markshannon