bbcmicrobit/micropython #281: Neopixel protocol flags

cefn, 2016-05-23:
Thanks for all the work on the micropython runtime.

I would like to port Neopixel logic already running on the Arduino-compatible Shrimp (e.g. https://twitter.com/ShrimpingIt/status/734834598039982081 running on  http://start.shrimping.it/kit/alarmclock.html ) but I'm struggling to know how to switch to the different KHZ required for the different types of pixels available under the 'Neopixel' label. 

Is there a facility for this in the library, or is it only WS2812 support?

Using the Adafruit library we can currently drive PL9823 with the flags...
NEO_GRB + NEO_KHZ800
...and the WS2811 modules with...
NEO_RGB + NEO_KHZ400
...as in the Word Clock.

See...
https://github.com/ShrimpingIt/projects/blob/master/sketchbook/shrimpingit/alarmclock/RGBUnaryClock/RGBUnaryClock.ino
...and...
https://github.com/ShrimpingIt/projects/blob/master/sketchbook/shrimpingit/alarmclock/WordClock/WordClock.ino
...for the full source.

dpgeorge, 2016-05-23:
The NeoPixel driver only supports a single frequency at the moment, and GRB byte order, to control WS2812 LEDs.  It's possible to improve the code to support other configurations.

cefn, 2016-05-24:
If this is ever in scope, the repository at https://github.com/adafruit/Adafruit_NeoPixel could be a useful reference for the variety of configurations which could be supported.