bbcmicrobit/micropython #284: Support for scroll_async

cefn, 2016-05-28:
The Microbit runtime has an asynchronous scroll behaviour which would be extremely useful for event-driven code, where blocking the liveness of the Microbit event loop just to display status is a real obstacle...

https://lancaster-university.github.io/microbit-docs/ubit/display/#scrollasync

Any chance of this making it into the microbit python API?

ntoll, 2016-05-28:
Hi @cefn, we don't use the DAL for the `display` module. If you want things to work in a non-blocking manner the convention is to pass in a `wait=False` named argument into the method. It's all explained in the API documentation here: http://microbit-micropython.readthedocs.org/en/latest/

(Specifically for the `display` module, here: https://microbit-micropython.readthedocs.io/en/latest/display.html)

Hope this helps!

cefn, 2016-05-28:
Thanks for the prompt response!

I'm not sure what you mean by 'the method'. There are at least two you might mean. scroll() or show()

scroll() seems to be for text scrolling (and doesn't apparently have a wait argument according to  https://microbit-micropython.readthedocs.io/en/latest/display.html). 

On closer reading, the second signature of show() permits letters to be passed in the iterable, so that will do for now although scroll would be better.

If scroll supports the wait argument, then it would be good to note that in the documentation.

ntoll, 2016-05-28:
Hi, I meant the `scroll` method, it does take the `wait` keyword, so the documentation is wrong. Nice find. :-)

ntoll, 2016-05-28:
Actually, the documentation IS correct, but it's not been rebuilt on readthedocs recently. Try again in 5 minutes and you should see the correct method signature for the `scroll` method. :-)

ntoll, 2016-05-28:
OK... it's corrected. :-)

markshannon, 2016-06-02:
Documentation now includes `wait`.

closed 2016-06-02 by markshannon