bbcmicrobit/micropython #286: Intermittent Nondeterministic Crash

cefn, 2016-05-29:
I have UART-reading code which runs correctly for quite a few iterations, but subsequently the Microbit appears to crash, with neither the 'success' logic or the 'failure' logic being executed and the display going dark. 

```
from microbit import *

uart.init(baudrate=9600,tx=pin4,rx=pin5)

def measure():
    uart.write(bytearray([0xFF, 0x01, 0x86, 0x00, 0x00, 0x00, 0x00, 0x00, 0x79]))
    gasdata = bytearray(9)
    if uart.readinto(gasdata)==9 and gasdata[1]==0x86:
        highppm = gasdata[2]
        lowppm = gasdata[3]
        ppm =  (highppm * 256) + lowppm 
        temp = gasdata[4] - 40
        return (ppm,temp)
    else:
        return None

while True:
    sample = measure()
    if sample == None:
        display.scroll("?")
    else: 
        display.scroll(str(sample[0]))
```

The script gathers data from a MH-Z16 Grove CO2 Sensor (Seeed) which provides a 3.3V UART output as described at http://www.seeedstudio.com/wiki/images/c/ca/MH-Z16_CO2_datasheet_EN.pdf

It was able to successfully report the PPM values expected for between 0 and 20 iterations or so before crashing. Then 0->10 iterations, then 0->2 iterations and now finally it doesn't seem to be willing to run a single iteration before crashing. As soon as the sensor device is plugged in, and hence it is starting to read the UART, it crashes.

I have been logging from this sensor over a 3.3V UART all afternoon to my desktop apparently without any issues, and I will resume testing this way to be sure the sensor itself hasn't developed any issues. However, I don't really understand how the sensor itself could be responsible for this crash  in any case.

Wiring is as simple as a common ground plus wires connecting TX and RX between the sensor and the declared pins in the micropython script (the CO2 sensor is running from its own independent 5V switched mains power supply).

## RECOVERING

To resume the original programmed behaviour...

* ...previously a push of the reset button on the back would sometimes resume the program, but this doesn't seem to be the case any longer
* ...if this didn't work, then was I uploading a Hello World program to it, then later uploading the UART-reading program, and that would give me a few more run cycles before another crash.
* ...now there seems to be no way to get a successful upload to run

Is there any way that this can be my python code? Is there any other tweaks worth trying as I believed I was just using a mainstream capability of the board.

dpgeorge, 2016-05-29:
If reset (or power cycle) doesn't fix it then it's not the Python that's the problem.

Note that pin4 is used to control the display, and pin5 is used for button A (see the docs).  So maybe you have a clash with the display and the uart that's causing the issue.  Try using pin13 and pin14 (or other unused pins).

Also, your code would be more efficient if you use:
```
uart.write(b'\xFF\x01\x86\x00\x00\x00\x00\x00\x79')
```

cefn, 2016-05-29:
I switched to using pin14 and pin15 as per https://github.com/bbcmicrobit/micropython/issues/287

This morning the Microbit connected using pin14 and pin15 and reported the CO2 ppm for several cycles, before crashing and requiring a power cycle.

Bizarrely, unplugging it from its USB power source altogether suddenly brought the whole thing back to life and it proceeded to successfully report PPM apparently using parasitic power from its RX pin or some stored power somewhere (lights were bright alarmingly, given it might have been drawing the power through TX from the sensor).

Replugged it and so far it's completed maybe 100 cycles, so fingers crossed. I will change the code for efficiency as you suggest.

I've had a go at authoring directly on the DAL through the mbed compiler, so if worst comes to worst hopefully I can produce a test case there which doesn't depend on python anyway.

I'm not crazy to think that the 3.3V UART documented at http://www.seeedstudio.com/wiki/images/c/ca/MH-Z16_CO2_datasheet_EN.pdf should just work, then?

dpgeorge, 2016-05-29:
The gas sensor should work.  Using the UART as you are using it is a normal thing to do.

Did you also connect the GND pins of the microbit and the sensor?

cefn, 2016-05-29:
Yes, I mentioned I wired a "common ground" in the original issue, but thanks for checking.

cefn, 2016-05-29:
Eventual crash after 250 cycles or so, resumed on reset for another 5 cycles, now locked up and not responding to reset button at all when wired to sensor TX and RX (pin14 and pin15). 

If sensor TX and RX are unwired, then Microbit can be reset and runs until TX and RX are reattached at which point it freezes and the display goes dark.

If the sensor is depowered, then Microbit can be reset and runs until sensor is powered on and two or three cycles have been completed, at which point the Microbit freezes again.

dpgeorge, 2016-05-29:
Very interesting.  Difficult to reproduce without having the same sensor.  If you can try implementing the same algo in C/C++ using mbed/DAL then that would allow to narrow down the bug.

cefn, 2016-05-29:
Problem remains across two different Microbits and with 1kOhm series resistors between TX and RX. I'm beginning to think it's a DAL liveness issue. Not sure how easy I'll find it to reliably replicate the code in DAL form, though.

cefn, 2016-05-31:
The nature of the hang has been tracked down, and we've managed to resolve it building on the DAL directly in the MBed compiler, thanks muchly to @jamesadevine . 

The major issue seems to relate to the special characters in the protocol of the MH-Z16 which need to be represented verbatim, even though some layers of the stack might consider them to be null-terminators.

However, it's impractical for me to guide learners through using the mbed toolchain. Is it reasonable to believe I was encountering a similar issue associated with the python serial implementation? (although not linked to ManagedString perhaps).

The presence of null filler characters in some frame-oriented protocols like this suggests the need for access to low level byte operations for reading and writing individual characters, assuming that the buffer-oriented operations are 'smart' about null terminators. 

If byte-oriented operations were available on the UART I imagine I could work around any problems with the buffer-oriented operations. Is this a possibility?

dpgeorge, 2016-05-31:
MicroPython does not use the DAL serial read/write functions.  It uses uBit.serial.getc() and uBit.serial.putc() to get/put individual characters.  So it doesn't have a problem with null chars, it just sends and receives byte objects.

DavidWhaleMEF, 2018-12-13:
@cefn There were a number of bug fixes and changes to get to V1.0.0 of MicroPython, many that happened in the uart area. Can you confirm this fixes your issue and/or can we close this ticket now?

cefn, 2018-12-13:
Unfortunately the device (with sensor and spare) was dispatched to the customer and not returned as expected. I can't easily recreate this now, without buying another £80 gas sensor, sorry.

closed 2018-12-13 by cefn