bbcmicrobit/micropython #287: Serial initialisation should throw error on invalid pins

cefn, 2016-05-29:
In an example program, the uart.init function accepted pin4 and pin5 as tx and rx arguments, although within http://m.pxt.io and noted at https://github.com/lancaster-university/microbit-docs/issues/22 there looks to be a limitation on which pins can be used for Serial.

If there is a limitation on valid Serial pins, the use of non-approved pins should cause an error to be thrown.

dpgeorge, 2016-05-29:
You can use pin4 and/or pin5.  But not at the same time as the display.  This is something better suited to describing in the docs rather than raising an error.

microbit-mark, 2018-10-26:
We need to think about where we describe this. We describe the method of turning the screen off to allow users to makes use of the pins here:
https://microbit-micropython.readthedocs.io/en/latest/display.html#microbit.display.off

but you're unlikely to read that as you are using UART for example.

Also, remapping pins should give pin mode errors if we are unable to use them.

https://github.com/bbcmicrobit/micropython/blob/3eaa746f98a07e09763a61cc707251020525fa4a/source/microbit/microbitpinmode.c#L97

We should document this a bit more in the micro:bit KB &/ Tech site

microbit-carlos, 2018-10-29:
We'll add some info about the display pins to https://microbit-micropython.readthedocs.io/en/latest/pin.html.

We tested what happens if display pins (in this case 4 and 6) are used for UART without disabling the display and while the display might not show images correctly, simple UART comms between 2 micro:bit still works as expected. So as Damien mentioned, this is better described in the docs rather than raising an error.

microbit-carlos, 2018-10-31:
PR https://github.com/bbcmicrobit/micropython/pull/589 has been merged with info about disabling the display to use those pins, so I'll close this issue.
Thanks everybody, and please feel free to reopen it if you think there if the issue is not completely resolved.

closed 2018-10-31 by microbit-carlos