bbcmicrobit/micropython #301: Setup Guide for build environment on Debian 8 (Jessie)

c-mart, 2016-06-04:
I had to jump around the docs and ask others for help, and from this experience, I produced the following guide. The existing "Debian and Ubuntu" section of the installation guide specifies adding new APT repositories but I found this wasn't necessary for Debian, everything is in stable/main.

I feel there is no problem keeping a set of distro-specific setup guides, as long as someone is willing to maintain each guide, which I'm happy to do for Debian. I would break up the "Debian and Ubuntu" section of the installation docs into a separate section for each distro. (I would also work some of this into flashfirmware.rst and .)

What do others think, would these contributions be welcome?

----------------------------------------------------

# Setup Guide: micro:bit MicroPython build environment on Debian 8 (Jessie)

This will allow you to build your own firmware and play with new features (like the radio module) which aren't in the official firmware yet. You don't need to do this if you only want to run Python code using the official firmware - you can just use Mu or uflash.

## APT Packages
Install these packages by running `apt-get install` with root privileges. All packages should be available in the stable/main repository.

- binutils-arm-none-eabi
- gcc-arm-none-eabi
- libnewlib-arm-none-eabi
- libnewlib-dev
- libstdc++-arm-none-eabi-newlib
- srecord
- cmake
- ninja-build

## Python Environment
Create a virtualenv and activate it:

    virtualenv microbit-micropython-venv
    source microbit-micropython-venv/bin/activate`

Install yotta and uflash:

    pip install yotta
    pip install uflash

## Get Repository
This example uses the official bbcmicrobit repo, but you can substitute this with another.
`git clone https://github.com/bbcmicrobit/micropython.git`

## Building firmware
`cd` to the root of the repository and run `yt target bbc-microbit-classic-gcc-nosd`. This will prompt you to sign up for an ARM mbed account, and you will need to confirm your email address.

After you sign up, run `yt target bbc-microbit-classic-gcc-nosd` again, log in with your browser when prompted. Address any error messages or missing dependencies.

Update yotta and run the build:

    yt up
    yt build

If all goes well, your firmware .hex file will be saved to `build/bbc-microbit-classic-gcc-nosd/source/microbit-micropython.hex`.

## Flashing firmware
Plug in your micro:bit. It should be mounted as a mass storage device.

### Using uflash
uflash allows you to flash a micro:bit with custom firmware and a Python script.

General usage:

    uflash -r [path to firmware] [path to Python file] [path to micro:bit mass storage volume]

Example:

    uflash -r build/bbc-microbit-classic-gcc-nosd/source/microbit-micropython.hex examples/space_shoot_multiplayer.py /home/yourname/media/MICROBIT

### By copying a .hex file
If you only wish to flash your firmware (and not also pass a Python file), you can just copy the firmware .hex to the micro:bit's mounted USB mass storage device.

Simply copy `/build/bbc-microbit-classic-gcc-nosd/source/microbit-micropython.hex` to the root directory of the MICROBIT volume.

Whether you use uflash or copy a .hex file, the firmware will flash to the microbit, yellow LED will blink for a few seconds, and then the micro:bit will boot.

c-mart, 2016-06-05:
Just tested this guide by setting up another PC with a fresh Debian installation, made several tweaks. I am now confident that this guide works for Debian 8.

ntoll, 2016-06-06:
I'll check all this with latest 16.04 LTS. I don't mind supporting Ubuntu either. I'll work out where's best to put this sort of documentation in our current docs sub-directory. There probably needs to be pages for Windows, OSX and Redhat/Fedora too.

bfayers, 2016-07-11:
After I flash my built firmware, is there anyway to have it stick, without having to use makecombinedhex.py or uflash? otherwise it just acts like a  regular program where when flashing something new it will overwrite it and go back to stock firmware

ntoll, 2016-07-11:
You can use uflash to use your custom firmware. See the bit about the `-r` / `--runtime` option here: https://uflash.readthedocs.io/en/latest/  e.g.

```
uflash -r custom_runtime.hex my_python_script.py
```

bfayers, 2016-07-11:
I have done this, i just wanted a slightly easier way of getting ode on without uflash etc, I found how anyway simply just flashed my firmware then used ufs put to put main.py and it keeps the python version + code

nickbailey, 2016-11-21:
I followed the instructions you gave and everything runs fine right up until the yt build. The C++ compilations fail with
```
In file included from /home/nick/Microbit/micropython/yotta_modules/mbed-classic/api/BusIn.h:19:0,
                 from /home/nick/Microbit/micropython/yotta_modules/mbed-classic/common/BusIn.cpp:16:
/home/nick/Microbit/micropython/yotta_modules/mbed-classic/api/platform.h:25:19: fatal error: cstddef: No such file or directory
compilation terminated.
```
I've seen this reported elsewhere, but it seems to date back to an old version of the C compiler in use with some distributions. Fixes involved symlinking version-specific include directories so they don't apply to me as I'm up-to-date AFAIK (Debian stretch; packages required by your guide from sid). Here's my module.json:
```
{
  "name": "microbit-micropython",
  "version": "0.0.1",
  "license": "MIT",
  "description": "MicroPython component for the BBC micro:bit",
  "keywords": ["mbed-classic", "microbit", "runtime", "library", "micropython"],
  "author": "Damien P George <damien.p.george@gmail.com>",
  "homepage": "http://micropython.org",
  "dependencies":{
    "mbed-classic": "~0.0.4",
    "microbit-dal":"lancaster-university/microbit-dal#v1.4.19b"
  },
  "extraIncludes":[
    "inc",
    "inc/lib",
    "inc/microbit"
  ],
  "bin": "./source"
}
```
and I attache the build log. I can't see anything obviously wrong, so I'm going to sleep on it and have another go tomorrow.
[log.txt](https://github.com/bbcmicrobit/micropython/files/604246/log.txt)

kjcole, 2021-02-08:
The `firmware-microbit-micropython` Ubuntu package contains `/usr/share/firmware-microbit-micropython/firmware.hex` but apparently that is not quite right? I will try reaching out to the Ubuntu gang that packages it, but just for a data point here:

```
Package: firmware-microbit-micropython
Version: 1.0.1-1
Installed-Size: 732
Maintainer: Python Applications Packaging Team <python-apps-team@lists.alioth.debian.org>
Architecture: all
Suggests: firmware-microbit-micropython-doc, picocom | screen
Enhances: mu-editor, python3-uflash
Conflicts: firmware-microbit-micropython-dl
Description-en: MicroPython runtime for the BBC micro:bit
 This package provides a binary firmware file for the  BBC micro:bit
 small board computer (SBC), containing the MicroPython runtime.
 .
 It is suggested that a dedicated flashing tool (e.g. uflash) is used to
 upload the firmware and Python scripts to the micro:bit device.
 .
 The uflash utility (package: python3-uflash) uses the MicroPython runtime
 provided in this package as the default firmware to flash to the micro:bit.
 .
 This package contains the MicroPython runtime for the BBC micro:bit
Description-md5: df6e22bc90b84c449f3657f35bb30f52
Homepage: https://github.com/bbcmicrobit/micropython
Section: python
Priority: optional
Filename: pool/main/f/firmware-microbit-micropython/firmware-microbit-micropython_1.0.1-1_all.deb
Size: 216240
MD5sum: 052afefb10bc8310766f7752a4572f9a
SHA256: 0c977bd81e284e966ae00d1d17759d346a0bc304d90eb240792acf6e28faceae
```