bbcmicrobit/micropython #313: Add information about usb mass storage mode to storage doc

mavhc, 2016-06-21:
https://github.com/bbcmicrobit/micropython/blob/master/docs/tutorials/storage.rst does not mention the usb mass storage mode that most people will be familiar with.

dpgeorge, 2016-06-24:
The mass storage mode is only for copying across new firmware, not storing files.  What sort of statements do you suggest adding to the docs?

ntoll, 2016-06-24:
I'm guessing that you shouldn't confuse the fs with mass-storage mode.

mavhc, 2016-06-24:
Yes, that is what I intended, to avoid confusion from people moving from
the mass storage mode of reprogramming the micro:bit to this more
sophisticated method. I'd expect most people to be familiar with the mass
storage mode and have that in mind as they were reading it. As it's a
tutorial I suggest:

Explaining how the fake mass storage system works, and how you'll have to
ignore that method and switch to the command line if you want to keep and
access files.

a) Noting that you need to flash it with micropython once.
b) Put the section about flashing it will wipe all your files in this
section
c) Explain how to copy a new main.py program to the microbit

Basically introduce the new workflow for those moving from drag and drop
via the website, then to using mu and clicking Flash, and now the command
line tools.

microbit-carlos, 2018-10-29:
We've added a note to clarify the difference between MicroPython file system vs. the micro:bit mass storage device in https://github.com/bbcmicrobit/micropython/pull/585.

closed 2018-10-29 by microbit-carlos