bbcmicrobit/micropython #328: pin19.write_digital(0) has no effect

deshipu, 2016-07-22:
I just tried to use I²C on a microbit with the new firmware (using mu), and I keep getting error -1010, and I can see on the osciloscope that the SCL pin is not pulsed at all. So I decided to see if the pins work at all, and it seems they are permanently stuck in high state.

The following code:
```
import microbit
microbit.pin19.write_digital(0)
microbit.pin20.write_digital(1)
```
Is supposed to bring the pins low, but it does not -- they remain high.

markshannon, 2016-07-23:
This is somewhat strange, and unfortunate.
Is this behaviour immediately after a reset, or does some other code need to run first?

deshipu, 2016-07-23:
It happened to me both just after reset and later, however, I am at a conference now and I'm using a little improvised way of accessing the microbit's pins (basically this: https://hackaday.io/project/9685-microheader) and I just realized that the pins might have been badly aligned and perhaps touching the 3V pin (they are right next to it). When I'm back home at Monday I will test this properly. I'm sorry, I should have waited with this bug report. I guess I became frustrated that I can't show the i2c capabilities of the microbit at the conference.

deshipu, 2016-07-30:
I finally took the time to repeat it, and still couldn't get the i2c to work. Thinking that it's a problem with contact with the pads, I tightened the connections and in the process I managed to damage the resistors close to the accelerometer. Now it always reads 0, and I don't think I will manage to get any i2c communications working on this particular unit.

I would love to hear from anyone who got the i2c to work on a recent version, though.

deshipu, 2016-09-15:
I got the kitronik edge connector, and confirmed that all my problems with pins were due to inadequate connections. I got an Adafruit LED backpack to work fine over I2C on the microbit. I'm closing this issue, sorry for the noise.

closed 2016-09-15 by deshipu