bbcmicrobit/micropython #331: Strange behaviour of pins when switching modes

markshannon, 2016-07-24:
The various behaviours of pins interact in strange and non-obvious ways.

For example, playing audio on a pin that is already set to an analog value temporary controls the pin, then reverts to the analog value, but writing a digital value is effectively ignored (the value only persists as long of the analog period)
When you start considering that some pins are dedicated to i2c and buttons, things start getting a bit weird.
You might want to play audio on the i2c pins, but you really ought to explictly disable i2c first.
Writing digital values to the button pins has a strange effect:
pin5.write_digital(1) and button A seems to function correctly,
but pin5.write_digital(0) and button A claims to be always pressed.

And as a really strange example:
```python
>>> pin5.set_analog_period(37)
>>> pin5.write_analog(512)
```
button_a flips between pressed and not pressed about once a second as the debouncing moves in and out of phase with the analog output.

What I would propose is:
* Button pins should be read-only
* The simpler modes, touch, digital, and analog, should be implicitly switched when a relevant method is called. This is supposed to happen already, but doesn't always.
* The more complex modes, audio, display, i2c, and spi, should need to be explicitly turned off before their pins can be used for anything else. 

So, for example, any attempt to read or write to one of the display pins should raise an exception unless the display has been turned off.

dpgeorge, 2016-07-25:
Isn't this an issue for upstream DAL?

markshannon, 2016-07-26:
It would be if didn't we implement our own display, pwm and audio modules.

markshannon, 2016-11-13:
Fixed by #345

closed 2016-11-13 by markshannon