bbcmicrobit/micropython #333: REPL seems not to work

muellmatto, 2016-08-01:
Connecting with screen or picocom from archlinux or macos doesn't show a prompt.
Typing makes the led flash, but it does not respond, neither does "pyboard.py" work. I flashed the latest Firmware from here ( https://www.mbed.com/en/development/hardware/prototyping-production/daplink/daplink-on-kl26z/#Updating_your_DAPLink_firmware ) but that doesn't solve the problem.

ntoll, 2016-08-01:
Have you flashed MicroPython onto the micro:bit. If there's no MicroPython on the device there's not going to be a REPL. (Just checking the obvious).

ntoll, 2016-08-01:
Why not `pip install uflash` and type `uflash` at the prompt with the micro:bit plugged in and mounted? That'll put MicroPython onto the device. Then try to connect as per the instructions in the README for this repos. Alternatively, just use Mu (http://codewith.mu/)

muellmatto, 2016-08-01:
Oh, i haven't thought about that one. I thougt, micropython is alway present. Thanks a lot.

closed 2016-08-01 by muellmatto