bbcmicrobit/micropython #334: ADC pins 4 and 10?

michael-freeman, 2016-08-05:
Building a small robot for schools. Trying to use ADC inputs on pins 4 & 10, but not working, guessing as these are shared with the display. Is this a fundamental hardware issue or could these pins be configured in software as just analogue inputs? Currently have two LDR potential dividers producing analogue voltages that vary from 0.75V to 2.5V, using the following code to read

def get_LDR1():
    tmp = pin4.read_analog()
    return tmp

def get_LDR2():
    tmp = pin10.read_analog()
    return tmp

When i connect these signals to these inputs the voltage is raised to +3V, adc conversion does work, but ive lost the signals i'm looking for.

dpgeorge, 2016-08-08:
Yes the display uses pins 4 and 10.  Try executing `display.off()` before doing the read_analog().

markshannon, 2016-08-09:
Also see #331

michael-freeman, 2016-08-11:
Cheers for the advice using display.off() definitely helps, but it looks like the LED is still loading the LDR circuit i.e. works fine on pin0, on pin4 readings will not go below 700. Playing around using a low impedance driver i.e. an opamp follower does seem to 'fix' the issue, but not knowing the microbits circuit diagram, perhaps a little dodgy. If you would like to see the robot/schematics/pcb i can send you a link, otherwise thanks for the help, job done :)