bbcmicrobit/micropython #335: Using music.play after speech.say results in no sound or garbled sound

iansparks, 2016-08-07:
Using the mu editor for python / microbit. It reports: MicroPython v1.7-9-gbe020eb

The following program:

```python
from microbit import *
import music
import speech
music.play(music.WAWAWAWAA, wait=True)
speech.say('Hello world')
sleep(2000)
music.play(music.WAWAWAWAA, wait=True)
```
Does not play the second WAWAWA... tune (with speaker attached to GND + PIN0). Remove speech.say() and it works as expected. Seems that speech.say leaves the microbit sound system in some altered state? I have tried music.reset() but does not seem to help.

dpgeorge, 2016-08-08:
The implementation of the music module was done first using simple PWM output.  Speech was added later uses the new audio module which can handle more sophisticated sounds and uses a more direct (low-level) way of outputing the data to the GPIO pin.  These 2 methods are not compatible.  Probably music should be updated to work alongside the newer audio functionality.

deshipu, 2016-08-08:
Alternatively speech should be updated to clean up after itself properly.

markshannon, 2016-08-08:
@deshipu Could you be more specific? What state should be restored?

deshipu, 2016-08-08:
I really don't know the specifics of how this works. All I know is that pin.write_analog() doesn't work after using speech/music on it, as we discovered recently on Europython with @ntoll.

markshannon, 2016-08-09:
`pin.write_analog()` uses the new PWM and is unrelated to`music.play()` which uses the old DAL version and, as @dpgeorge says, needs to be updated.

@deshipu Could you please file a separate issue if `pin.write_analog()` doesn't work properly.

deshipu, 2016-08-09:
I see, sorry, it sounded similar to me. I will file an issue.

Tonyaj, 2016-10-24:
Hi
I have been having the same problem using speech and music in same program.
Couldn’t see what I was doing wrong, then someone kindly pointed this out.
Is there any update or solution ?
Thanks

PR #345 (2016-11-12): Safer behaviour when the mode of a pin changes. Fix up PWM implementation
Rework the pwm module to allow smooth transitions when period or duty cycles are changed.
Make music module use our pwm (not the DAL's pwm).
Modify the pin and related modules to ensure that attempting to use a pin when it is already used for another purpose, raises an exception.

For example:

``` python
>>> pin5.write_analog(300)
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
TypeError: Pin 5 in button mode
>>> pin6.write_analog(300)
Traceback (most recent call last):
  File "<stdin>", line 1, [... cut]
closed 2016-11-12 by dpgeorge