bbcmicrobit/micropython #336: Communication to PC over USB?

MikeDX, 2016-08-08:
In addition to the REPL interface, is there any way to communicate with the micro:bit over usb in code? I couldn't see any, short of making up a secondary serial via the pins.

Is this possible at all? I've tried RTFM but to no evail.

I currently have a communication to the PC using an intermediate lib to call REPL functions one at a time, but that's not ideal - I'd rather have a python script running on the device that handles that in real time rather than individual python function calls over REPL.

Apologies for using the issue tracker to ask this, but if it isn't something that's currently supported, maybe this could be a future feature?

dpgeorge, 2016-08-09:
You can use the "uart" object to read/write over the USB serial connection,
eg:

>>> uart.write('hello!')

MikeDX, 2016-08-09:
Thanks @dpgeorge, I didn't realise you could use the uart in that way - it was my understanding the uart had to be initialised somehow but just using uart.read/write works exactly how I want.

closed 2016-08-09 by MikeDX