bbcmicrobit/micropython #337: pin0.write_analog stops working after music.play()

deshipu, 2016-08-09:
After the following code:
```
>>> from microbit import *
>>> import music
>>> pin0.set_analog_period(50)
>>> pin0.set_analog_period(20)
>>> pin0.write_analog(512)
```
I get the following result:
[image]

Then after this:
```
>>> music.play(music.WAWAWAWAA)
>>> pin0.write_analog(512)     
```

I get no output on pin0:
[image]

deshipu, 2016-08-09:
Using ``speech.say()`` in place of ``music.play()`` doesn't have that effect.

carlosperate, 2018-03-20:
I've just tested this with a speaker and oscilloscope with version 0.9 (the one on the online editor since October 2016) and  while #345 did fix this in a way (`pin0.write_analog()` works fine after `music.play()`), something has caused the `music.play(music.WAWAWAWAA)` to not reproduce the tune output on pin0, basically the first `pin0.write_analog()` continues uninterrupted.

I've tested the same with the latest master (with the version1 branch merged) and I can confirm this version works as expected:
*  `pin0.write_analog()` produces the correct output
* `music.play(music.WAWAWAWAA)` plays the tune as expected
* second `pin0.write_analog()` works again, however, this second `write_analog` uses the period last configured by the music module.

We haven't really defined any default period for the `write_analog` so I would consider this expected behaviour.

Both issues are fixed on master, so I'll close this one.

closed 2018-03-20 by microbit-carlos