bbcmicrobit/micropython #338: Firmware not compiling using yotta under Windows 10 (ninja)

AlexisTM, 2016-08-14:
Dear,

The actual master do not compile using Yotta.

System : 

- commit : bfaa52a2bb1c934a89202d44bb50172405a5b1bb 
- OS : Windows 10
- Yotta  : v023

Commands : 

```
$ git clone https://github.com/bbcmicrobit/micropython
$ cd micropython
$ yt target bbc-microbit-classic-gcc-nosd
$ yt up
$ yt build
```

Output : 

> (the long command)
> D:/Workspace/yotta/micropython/yotta_modules/ble-nrf51822/source/nRF5xGap.cpp: In member function 'virtual ble_error_t nRF5xGap::setWhitelist(const Gap::Whitelist_t&)':
> D:/Workspace/yotta/micropython/yotta_modules/ble-nrf51822/source/nRF5xGap.cpp:689:59: error: 'BLEProtocol::AddressType_t' is not a class or namespace
>          if (whitelistIn.addresses[i].type == BLEProtocol::AddressType_t::RANDOM_PRIVATE_NON_RESOLVABLE) {
>                                                            ^
> D:/Workspace/yotta/micropython/yotta_modules/ble-nrf51822/source/nRF5xGap.cpp: In member function 'ble_error_t nRF5xGap::generateStackWhitelist(ble_gap_whitelist_t&)':
> D:/Workspace/yotta/micropython/yotta_modules/ble-nrf51822/source/nRF5xGap.cpp:906:61: error: 'BLEProtocol::AddressType_t' is not a class or namespace
>          if (whitelistAddresses[i].addr_type == BLEProtocol::AddressType_t::RANDOM_PRIVATE_RESOLVABLE) {
>                                                              ^
> ninja: build stopped: subcommand failed.
> error: command ['ninja'] failed

chrizmo, 2016-08-15:
I am experiencing the same issue on Linux (Ubuntu 16.04.1):

Output:

> /microbit/micropython/yotta_modules/ble-nrf51822/source/nRF5xGap.cpp: In member function 'virtual ble_error_t nRF5xGap::setWhitelist(const Gap::Whitelist_t&)':
/microbit/micropython/yotta_modules/ble-nrf51822/source/nRF5xGap.cpp:689:59: error: 'BLEProtocol::AddressType_t' is not a class or namespace
         if (whitelistIn.addresses[i].type == BLEProtocol::AddressType_t::RANDOM_PRIVATE_NON_RESOLVABLE) {
                                                           ^
/microbit/micropython/yotta_modules/ble-nrf51822/source/nRF5xGap.cpp: In member function 'ble_error_t nRF5xGap::generateStackWhitelist(ble_gap_whitelist_t&)':
/microbit/micropython/yotta_modules/ble-nrf51822/source/nRF5xGap.cpp:906:61: error: 'BLEProtocol::AddressType_t' is not a class or namespace
         if (whitelistAddresses[i].addr_type == BLEProtocol::AddressType_t::RANDOM_PRIVATE_RESOLVABLE) {
                                                             ^

dpgeorge, 2016-08-15:
What version of gcc (arm-none-eabi-gcc)?

chrizmo, 2016-08-15:
> gcc version 5.4.0 20160609 (Ubuntu 5.4.0-6ubuntu1~16.04.2) 

>gcc-arm-none-eabi/xenial,now 15:4.9.3+svn231177-1 amd64 [installed]

AlexisTM, 2016-08-15:
I have an older version, the default one of Yotta : 

gcc version 4.9.3 20150529 (release) [ARM/embedded-4_9-branch revision 227977] (GNU Tools for ARM Embedded Processors)

dpgeorge, 2016-08-15:
The build works fine for me, with arm-none-eabi-gcc 6.1.1.

I think the problem is that the "old" gcc version you use can't handle the advanced C++ feature that's used by the code (BLEProtocol::AddressType_t is a typedef of an enum within a struct within a namespace).  We can't fix such problems here, so please report the bug upstream at the yotta module here: https://github.com/ARMmbed/ble-nrf51822

chrizmo, 2016-08-15:
Excuse my ignorance, but I have serious trouble obtaining arm-none-eabi-gcc 6.1.1 to confirm that it's a compiler version issue. Could you point me in the right direction?

I tried compiling with version 5-2016-q2-update of the toolchain and gcc 6.1.1 but experienced the same issues.

dpgeorge, 2016-08-16:
> Excuse my ignorance, but I have serious trouble obtaining arm-none-eabi-gcc 6.1.1 to confirm that it's a compiler version issue. Could you point me in the right direction?

I use Arch Linux and 6.1.1 is the latest version that it ships.  But if you tried 5-2016-q2-update then that's a pretty recent v5.x.x so should work.  This bug really needs to be reported upstream.

chrizmo, 2016-08-16:
I suspected you were using Arch. I'll report it upstream right away.

Xykon, 2016-08-27:
I've had the same problem... tried with arm-none-eabi-gcc versions 4.9.3 20150529 and 5.4.1 20160609.

The problem seems to be with lancaster-university/nrf51822#v2.5.0+mb5 which is used by lancaster-university/microbit-dal#v1.4.19

To make things easy I created my own version of microbit-dal to use nrf51822#v2.5.0+mb6 instead of mb5 which seems to work OK (I can build and run micropython though I haven't tested all functionality yet).

If you want to try this fix simply update module.json in the main micropython directory and replace:

`"microbit-dal":"lancaster-university/microbit-dal#v1.4.19"`

with

`"microbit-dal":"Xykon/microbit-dal#v1.4.19-mb6"`

daitangio, 2016-09-08:
Hi, the simple fix suggested by Xykon worked under MacOSX too. I need to install thus version of the compiler:https://launchpad.net/gcc-arm-embedded/+download
because the macports was bugged too
(see 
http://stackoverflow.com/questions/26931979/gnu-arm-nano-specs-not-found)

zeth, 2016-09-19:
I may be stating the obvious but I had the same issue on Debian Stretch and the same work around from Xykon made it build.

carlosperate, 2016-10-21:
Raised https://github.com/lancaster-university/microbit-dal/issues/228, so we could fix this by updating the DAL version dependency.

PR #374 (2016-11-07): Update DAL to v1.4.19b.
This fixes a compile issue with the ble-nrf51822 component.

Will fix #338.
closed 2016-11-07 by markshannon