bbcmicrobit/micropython #339: storage.rst examples seem not to work as expected

markbasham, 2016-08-14:
in https://github.com/bbcmicrobit/micropython/blob/master/docs/tutorials/storage.rst there is a simple  example snippet of code (line 98 and 99)

    with open('hello.txt', 'w') as my_file:
        my_file.write("Hello, World!")

If this is put onto my microbit, I get the following error:

    line 1 AttributeError 'NoneType' object has no attribute '__exit__'

dpgeorge, 2016-08-15:
It works for me.  What version of MicroPython are you using (press ctrl-D to see)?

markbasham, 2016-08-16:
rookie comment, but ctrl-d where, I'm using the bbc microbit webpage to edit and upload, is there a better way?  Will have an investigate now.

markbasham, 2016-08-16:
So it works for me fine using 'mu' as an edtor and uploading from there, so im happy :) but if i put the same code in the online editor, donload it and copy the .hex file to the bbcmb then it fails with the message above.  Dont know if this is worth a mention in the doc?  Have left open if you want to do something, otherwise feel free to close.

Cheers

Mark

dpgeorge, 2016-08-17:
> but if i put the same code in the online editor,

The online editor uses an old version of the firmware which does not support files.

closed 2016-08-17 by dpgeorge

ntoll, 2016-08-17:
Hi @markbasham. You may ask why the online editor is not up-to-date (and be right to be annoyed that it isn't).

Unfortunately, this is something beyond out control. The BBC have final say over when the online editor is updated and we have a bunch of pending updates waiting for them to approve and generally move through their bureaucracy. We're waiting on them to give the go-ahead on one final feature (the radio module) before we'll be allowed to update the runtime.

I'm sorry I can't be more helpful with this, but the BBC are the blockers (although we [the Python community] are working on an alternative "work around" solution that'll be available online in the coming weeks) and our friends in the BBC are trying to push this forward in the face of big bureaucratic corp process.

markbasham, 2016-08-17:
No problem at all, the mu editor is much better than the online one, its just not quite as obvious when you start out.  Thanks for both your help anyhow :)