bbcmicrobit/micropython #341: Suggestions about connecting speakers (speech module)

whaleygeek, 2016-09-01:
This resource:

https://microbit-micropython.readthedocs.io/en/latest/tutorials/speech.html

...suggests connecting a speaker directly to the pins. It has caused a few questions on the micro-python mailing list about whether this could cause damage to the micro:bit.

There is a small chance that directly connecting a bare (wound) speaker to the pins of a micro:bit could cause back-emf from the inductive load, that exceeds the specified maximum pin voltage of the GPIO pins on the NRF51, causing permanent damage. Not knowing the specific make and model and specification of speaker, it is hard to say whether damage would be caused or not.

My recommendation would be to use a powered speaker. It might be worth in the meantime, to be on the safe side, changing the docs to suggest using a powered speaker.

However, some advice could be added to the docs, as to additional components to add to the speaker connection that would give suitable protection against over-voltage from inductive loads. For example, something like this response (from DarioG on 17 Nov 2007), the Zeners are for over voltage protection. http://www.microchip.com/forums/m295795.aspx

I haven't done the maths or taken any experimental readings, it might be that in practice the frequencies involved mean that the back-emf generated from the inductive load when it is being switched is still within the specified maximum working voltage of the pins.

I just wanted to put this issue here as a placeholder so we don't forget about this fact. If I get a chance I will test a few different arbitrary speakers and do some rudimentary maths to see how close to the voltage limit of the micro:bit pins it reaches.

mikethebee, 2016-09-09:
Thank you WhaleyGeek for the post. I have the same concerns, various docs are contradictory. I have an app that has limited power and so I am looking to use a passive speaker and passive amp (a horn) I tried 8ohm and 4ohm speakers, large and small, and they worked but I am concerned about current draw over a long period of use. I do plan to get help from our Maker group to do the tests you mention. -Mike

whaleygeek, 2016-09-09:
Thanks Mike, let us know what you discover with the Maker group.

There is sometimes a fine line between "works" and "works reliably for all
time", and often there are a lot of great learning opportunities in
'digging under the surface' a bit and understanding really what is going
on. I'd like to think that with the micro:bit we can encourage users to ask
those sorts of questions and really understand how things work and learn
new things as a result.