bbcmicrobit/micropython #343: range of accelerometer.get_x()

pintman, 2016-09-18:
What is the range of valies that ``accelerometer.get_x()`` can return? It is only mentioned that the values will be positive or negative. Please provide max and min values in the documentation.

whaleygeek, 2016-09-18:
By default it is +-2g*1000 on all 3 channels x y z but there is a little
bit of noise on that, and you can select wider ranges in the underlying C
runtime. Nominally +- 1000 for tilt and +-2000 if moving, although with a
bit of noise and integer rounding error, you should probably be careful to
clamp it in your code if min max are critical and would cause overflows or
underflows elsewhere as a result.

pintman, 2016-09-18:
Thanks for clarifying. :)

Will it be added to the official documentation? :)

whaleygeek, 2016-09-18:
Hi Marco,

Well, Nicholas's usual answer to this would be 'please do send in a pull
request with the suggested changes'.

I'm not sure that is *always* the right way to do it for everyone who might
raise a query about the docs!

Although I know for example that the authoritative information about the
accelerometer range came from jfinney at Lancaster Uni, and the specific
code that sets these limits is inside the Lancaster runtime source code,
which is hosted here, and in particular in this line in this file in the
revision that is current as of today (18th Sept 2016):

https://github.com/lancaster-university/microbit-dal/blob/
master/source/drivers/MicroBitAccelerometer.cpp#L268

In combination with reading the data sheet for the Freescale MMA8653 here:
http://cache.freescale.com/files/sensors/doc/data_sheet/MMA8653FC.pdf

It would be possible to correctly work out the maximum range of x y and z.

I note (which is consistent with what jfinney told me a few months ago)
that a scale factor of 1024 is applied to the incoming data at those lines.
There are some opportunities for noise and rounding etc.

On 3rd Nov 2015, this was what I got from jfinney when I asked about the
range, as I was at the time having strange overflow problems with code I
was developing:

"The default accelerometer range is +/-2g, but it is configurable. (+/- 1g,
2g, 4g, 8g). In any case, the value given is in absolute SI units –
milli-Newtons. There is a bit of a rounding issue in there at the moment
(it returns a range +/-2048 rather than +/-2000)...."

I believe the rounding issues were solved before the code was open sourced.

Also, the official runtime documentation for the getX() is here, it does
not state what the maximum return value is, but it does say it is in
milli-g.

https://lancaster-university.github.io/microbit-docs/ubit/accelerometer/

You might also like to issue a PR against the Lancaster Uni docs here, as
MicroPython basically sits on top of this and a lot of the initial text for
the MP docs was probably inspired by the Lancaster Uni DAL docs anyway..

https://github.com/lancaster-university/microbit-docs/blob/master/docs/ubit/accelerometer.md

So, there probably is enough data there in front of you to work out the
correct answer and issue a PR for a documentation update, if you are
feeling brave?

David

pintman, 2016-09-19:
That is a very detailed answer. Thank you. :) 

I have not enough experience to write this kind of documentation - at least not at the moment. However thank you for a deeper insight into this kind of measurements. I did not think it to be such complex. ;)

tjguk, 2016-09-19:
I'm just going to weigh in a little and, well, attempt to channel @ntoll who's currently at PyCon UK probably sprinting on microbit-related stuff. The suggestion to "send us a PR" is not intended as a dismissal, as if to say: "Put up or shut up!". Rather, it's intended to emphasise  how collaborative and open this process is: you don't have to wait for some staffer to update the docs -- you can do it yourself!

However, it's always acceptable either, as @pintman has done, to say "thanks for the info but I don't feel up to writing the docs myself" or to say, for example, "I don't really get how github works, but this is what I'd say here: .. .blah blah text blah". Someone can then pick that up and incorporate it into the text as they see fit.

Of course, it may be that no change to the docs is really suitable (which might be the case here; I'm not sure). In which case, the discussion is still valuable for clarifying something and its outcome might provide the basis for some update in the future.

ntoll, 2016-09-20:
What @tjguk said. I :heart: the Python community and its willingness to collaborate, support and be open at the same time.

@whaleygeek why not create a précis of your answer and put it into the appropriate place in the docs? PR most welcome (and you've written most of it already).

microbit-carlos, 2018-10-30:
The range and units for the accelerometer have been added via #584, thank you everybody!

closed 2018-10-30 by microbit-carlos