bbcmicrobit/micropython #347: Frozen Modules

simoncragg, 2016-09-22:
Hi there,

I am attempting to build my own `microbit-micropython.hex` file which includes a frozen module.  I am completely new to python and c++ so please accept my apologies if this is a really silly question :)

Starting with some positives, I am able to successfully build a `microbit-micropython.hex` file from the untouched source code using yotta.  I can also use the `makecombinedhex.py` tool to produce a running program on my bbc microbit.  Happy days!  However, my true goal is to add a frozen module to the build, to allow me to off-load some classes from my python program - to provide enough space for one or two additional features.

Here are the steps I have performed so far-

(EDIT: Platform Windows 10)

1.  Downloaded the make-frozen.py tool from the [micropython git repository](https://github.com/micropython/micropython) (note: not this repo).
2.  Used make-frozen.py tool to produce a very simple frozen module  (time.py -> time.c).
3.  Added the time.c file to the `/source/py` directory (not sure if this is correct).
4.  Set `#define MICROPY_MODULE_FROZEN       (1)`  in the `inc/microbit.mpconfigport.h` file.
5.  Ran the following yotta command:  `yt build`

Here is the linker error:-

```
Recompacting deps...
[2/2] Linking CXX executable source/microbit-micropython
FAILED: cmd.exe /C "cd . && C:\yotta\gcc\bin\arm-none-eabi-g++.exe   -fno-exceptions -fno-unwind-tables -Wl,--gc-sections -Wl,--sort-common -Wl,--sort-section=alignment -Wl,-wrap,main -mcpu=cortex-m0 -mthumb -T"C:/MicroPython/yotta_targets/bbc-microbit-classic-gcc-nosd/CMake/../ld/NRF51822.ld" -Wl,-Map,source/microbit-micropython.map -Wl,--start-group @CMakeFiles/microbit-micropython.rsp   -lstdc++ -lsupc++ -lm -lc -lgcc -lstdc++ -lsupc++ -lm -lc -lgcc -Wl,--end-group  --specs=nano.specs -o source/microbit-micropython && cmd.exe /C "cd /D C:/MicroPython/build/bbc-microbit-classic-gcc-nosd/source && arm-none-eabi-objcopy -O ihex microbit-micropython microbit-micropython.hex && srec_cat microbit-micropython.hex -intel -o microbit-micropython.hex -intel --line-length=44 && cd /D C:/MicroPython/build/bbc-microbit-classic-gcc-nosd/source && C:/yotta/gcc/bin/arm-none-eabi-objcopy.exe -O binary microbit-micropython microbit-micropython.bin""
source/CMakeFiles/microbit-micropython.dir/C_/MicroPython/source/py/frozenmod.c.o: In function `mp_find_frozen_module':
C:/MicroPython/source/py/frozenmod.c:59: undefined reference to `mp_frozen_names'
C:/MicroPython/source/py/frozenmod.c:59: undefined reference to `mp_frozen_content'
C:/MicroPython/source/py/frozenmod.c:59: undefined reference to `mp_frozen_sizes'
collect2.exe: error: ld returned 1 exit status
ninja: build stopped: subcommand failed.
error: command ['ninja'] failed
```

I can see that a `time.c.o` file has been created in the `build/bbc-microbit-classic-gcc-nosd/source/CMakeFiles/microbit-micropython.dir/C_/MicroPython/source/py` directory.  Looking at the source file (time.c), I can see that the `mp_frozen_names`, `mp_frozen_content`, and `mp_frozen_sizes` variables are all defined and populated.

Please could someone advise?  Maybe I should just drop the additional features :)

Many Thanks
Simon

dpgeorge, 2016-09-22:
You almost had it!  The problem is that the microbit uses v1.7 of upstream MicroPython, so you need to get the make-frozen.py file from that repo as at that version (ie git checkout v1.7, then copy make-frozen.py).

Note that this kind of frozen module will still take RAM to compile it.  You probably need frozen bytecode which requires a bit more effort (you use mpy-cross to cross compile bytecode to a .mpy file, then mpy-tool.py to turn the .mpy to a .c; see stmhal/Makefile for hints how to do this).

simoncragg, 2016-09-22:
Hi Damien,

Many thanks for your help!  The 1.7 version of the `make-frozen.py` tool did indeed work!! However, I'm now struggling to reference and use the frozen module from inside my python program.  Please could you advise?  Thanks also for pointing out that I actually need a frozen bytecode.  I will be persuing this whatever the outcome of my current issue.

```
import Time
from microbit import *

time = Time()
time.hours = 10
display.show("Hello there")
```

.. but I received the error` Line 1 ImportError no module named 'Time'`

I tried removing the `Import Time` line, but the program didn't seem to run.  There was no error message on the device however the "Hello there" message was not displayed.

The following code, did work:-
```
from microbit import *

display.show("Hello there")
```

Many Thanks
Simon

dpgeorge, 2016-09-23:
You need to do: `t = time.Time()`.  That's just how normal Python works.

simoncragg, 2016-09-23:
Oops :) Many thanks for pointing this out!  I should have spotted this mistake myself, but I'm new to the language.  Many thanks Damien for all your help with this issue!  I couldn't have got there without some help and I am extremely grateful for your time!  Moving forward, I'm going to follow your guidance and attempt to build a frozen bytecode module.  I'm going to close this ticket now though as my original issue have been resolved.

simoncragg, 2016-09-23:
On second thoughts ...  If it's acceptable, I'd like to keep this issue open as I will most likely need a little help in building a frozen bytecode module.  If I manage to do it myself, I'll be sure to update and close this issue.

simoncragg, 2016-09-23:
OK so I am now attempting to build a frozen bytecode module.  I believe the required steps are:-

1. Make `mpy-cross` from the source files found [here] (https://github.com/micropython/micropython/tree/v1.7/mpy-cross) (v1.7)
2. Run the `mpy-cross` program to cross-compile my .py file into a .mpy file.
3. Use the `mpy-tool.py` tool from [here] (https://github.com/micropython/micropython/tree/v1.8.4/tools) (v1.8.4) to turn the .mpy file into a .c file.

I am currently experiencing an issue with step 1:

Although I have previously been using Windows 10, the requirement to make the mpy-cross program has forced to move to OS X El Capitan (10.11.6).  Luckily I have a dual-boot Macbook so this was pretty straightforward :)  I downloaded the micropython-1.7 source code, opened up a terminal, navigated to the mpy-cross folder and typed `make`.  Things got off to a go start but then I was hit with an error.  Here is the full output:-

```
$ make
Use make V=1 or set BUILD_VERBOSE in your environment to increase build verbosity.
mkdir -p build/genhdr
GEN build/genhdr/qstrdefs.generated.h
Generating build/genhdr/mpversion.h
mkdir -p build/py
mkdir -p build/py/../extmod
CC ../py/mpstate.c
CC ../py/nlrx86.S
CC ../py/nlrx64.S
CC ../py/nlrthumb.S
CC ../py/nlrxtensa.S
CC ../py/nlrsetjmp.c
CC ../py/malloc.c
CC ../py/gc.c
CC ../py/qstr.c
CC ../py/vstr.c
CC ../py/mpprint.c
CC ../py/unicode.c
CC ../py/mpz.c
CC ../py/lexer.c
CC ../py/lexerstr.c
CC ../py/lexerunix.c
CC ../py/parse.c
CC ../py/scope.c
CC ../py/compile.c
CC ../py/emitcommon.c
CC ../py/emitbc.c
CC ../py/asmx64.c
CC ../py/emitnative.c
CC ../py/asmx86.c
CC ../py/emitnative.c
CC ../py/asmthumb.c
CC ../py/emitnative.c
CC ../py/emitinlinethumb.c
CC ../py/asmarm.c
CC ../py/emitnative.c
CC ../py/formatfloat.c
CC ../py/parsenumbase.c
CC ../py/parsenum.c
CC ../py/emitglue.c
CC ../py/runtime.c
CC ../py/nativeglue.c
CC ../py/stackctrl.c
CC ../py/argcheck.c
CC ../py/warning.c
CC ../py/map.c
CC ../py/obj.c
CC ../py/objarray.c
CC ../py/objattrtuple.c
CC ../py/objbool.c
CC ../py/objboundmeth.c
CC ../py/objcell.c
CC ../py/objclosure.c
CC ../py/objcomplex.c
CC ../py/objdict.c
CC ../py/objenumerate.c
CC ../py/objexcept.c
CC ../py/objfilter.c
CC ../py/objfloat.c
CC ../py/objfun.c
CC ../py/objgenerator.c
CC ../py/objgetitemiter.c
CC ../py/objint.c
CC ../py/objint_longlong.c
CC ../py/objint_mpz.c
CC ../py/objlist.c
../py/objlist.c:272:78: error: shifting a negative signed value is undefined
      [-Werror,-Wshift-negative-value]
  ...self->len, n_args == 1 ? MP_OBJ_NEW_SMALL_INT(-1) : args[1], false);
                              ^~~~~~~~~~~~~~~~~~~~~~~~
../py/obj.h:86:78: note: expanded from macro 'MP_OBJ_NEW_SMALL_INT'
#define MP_OBJ_NEW_SMALL_INT(small_int) ((mp_obj_t)((((mp_int_t)(small_int)) << ...
                                                     ~~~~~~~~~~~~~~~~~~~~~~~ ^
1 error generated.
make: *** [build/py/objlist.o] Error 1
```

Please can someone advise as to how I can overcome this barrier?

Many Thanks
Simon

simoncragg, 2016-09-23:
Hmm I've fixed my latest issue (`mpy-cross` make error) by using the source code from the master repo and not v1.7.

dpgeorge, 2016-09-23:
I don't think v1.7 supported frozen bytecode, and you won't be able to use mpy-cross/mpy-tool from master because the bytecode has changed since v1.7 and it'll generate incorrect code.  Sorry but there's really no easy way to get it working.

simoncragg, 2016-09-23:
Hi Damien,

So close yet so far away!  I was just about to try and build a `microbit-micropython.hex` file using my new frozen bytecode module.  I guess if the byte code has changed between these versions ,then there's no point trying.  

Would it be in anyway possible to introduce master's `mpy-cross` and `mpy-tool.py`files  into the bbcmicrobit / micropython repo and make them compile / work with that code base's version of the byte code?

dpgeorge, 2016-09-27:
Stripping out C-level modules won't help much to reduce RAM.  The microbit has very little RAM to begin with so there's only so much you can do with it.

The easiest way to get frozen bytecode on the microbit would be to upgrade the MicroPython core to the latest version.  That requires some effort/time from my side, in particular it depends on this issue: https://github.com/micropython/micropython/issues/2417

simoncragg, 2016-09-29:
Hi Damien, 

Many thanks for all your help on this post.  Please don't commit yourself on my account!  I'm sure you have your own motivations to improve the implementation and so if frozen bytecode support ever does come to fruition then I'll be sure to take advantage of it :)

I'll close this ticket now as this has become more of a feature request.

Many Thanks
Simon

closed 2016-09-29 by simoncragg