bbcmicrobit/micropython #348: problems with files

catchpolej, 2016-09-22:
I am a teacher interested in my students doing some accelerometer measurements.
I cannot get the file handling to work. For example:

from microbit import *

story_file = open('story.txt','r')
line = story_file.read(5)
print(line)
story_file.close()

Line 5 is the read() instruction and the microbit display gives me:
Error Line 5: Attribute error: Nonetype object has no attribute etc.....

Also if I......
import os:  I get a No such module eror

I am pulling the scripts down from the mictobit site python code editor page.

deshipu, 2016-09-22:
The version on the website is old, and doesn't have a filesystem yet. You will need to use a newer version, for instance the one shipped with the Mu editor.

ntoll, 2016-09-22:
You should use the Mu editor (http://codewith.mu/)

The BBC have still not signed off the latest version of the online editor and I'm powerless to update until I get their permission. :-(

I'm sorry about this.

However, there will be an alternative, updated and bug-fixed version of the editor outside the BBC's control launching soon.

catchpolej, 2016-09-22:
Thanks - trying it now but currently getting an error installing mu:
Error loading Python lib '/tmp/_MEII6NAKa/libpython3.4m.so.1.0': /lib/x86_64-linux-gnu/libc.so.6: version `GLIBC_2.17' not found (required by /tmp/_MEII6NAKa/libpython3.4m.so.1.0)
On Linux; I installed python 3.4 after I saw error first time (was on 3.3) but error persists.
..................................... How soon? ....... :-)

ntoll, 2016-09-22:
What version of Linux are you using? I suspect it may be as simple as you using a 32bit system and we've built for 64bit (hence the lack of finding the correct version of libc).

WRT the alternative editor - *very soon* (weeks). :-)

carlosperate, 2016-09-22:
@catchpolej If you were trying to run the Mu executable for Linux it is indeed made for 64bit OS. If you are running a Debian based distribution (like Ubuntu) you can still install it from one of the deb files created for the raspberry pi: https://s3-us-west-2.amazonaws.com/ardublockly-builds/microbit/raspberry_pi/mu-2016-09-20_15_27_27.deb

catchpolej, 2016-09-22:
I have updated to Python 3.4 successfully and I think I am running 64 bit:
uname -a
Linux jca-HP-255-G1-Notebook-PC 3.13.0-61-generic #100~precise1-Ubuntu SMP Wed Jul 29 12:06:40 UTC 2015 x86_64 x86_64 x86_64 GNU/Linux
but mu.bin installer is still giving error message.

carlosperate, 2016-09-22:
The `mu.bin` file is not an installer, but an executable for the Mu application that contains the Python runtime embedded. In this case it needs a version of glibc that it cannot find in your system. If you'd like to install the deb package instead it will fetch and install all the dependencies to run Mu from your local Python installation:
https://s3-us-west-2.amazonaws.com/ardublockly-builds/microbit/raspberry_pi/mu-2016-09-20_15_27_27.deb
```
> sudo dpkg -i mu-2016-09-20_15_27_27.deb
> sudo apt-get -f install
```

catchpolej, 2016-09-22:
The mu.........deb file opened Ubuntu Sofware centre but came back with "Dependancy is not satisfiable: python3(>=3.3.2-2~)"
The dpkg command complained about many python components not being there so did not complete. (Though I have just updated python to 3.4) !?

carlosperate, 2016-09-22:
Since this conversation is not relevant to micropython any more, perhaps we should move this conversation to the Mu repository? If you open an issue there I'd be able to give you a bit more support.
https://github.com/mu-editor/mu/issues/

catchpolej, 2016-09-22:
Yes; I will open an issue tomorrow in the Mu repository if I cannot work out a solution.

catchpolej, 2016-09-26:
The Mu issue is now resolved and file writing/reading is working. Next stage is to get the file data off the microbit onto a computer......I am trying to use microfs but though pip installation of it claims to be successful python is claiming no module microfs....is this issue resolved anywhere?

ntoll, 2016-09-26:
You're following the instructions here..?

https://microfs.readthedocs.io/en/latest/

catchpolej, 2016-09-26:
Yes I have followed these instructions from Linux command line
pip install microfs worked
although
pip install --no-cache --upgrade microfs
gives no such option -n
in python programs no such module microfs and command line ufs is not defined

carlosperate, 2016-09-26:
Are you using Python 3? keep in mind that `pip` probably install the modules in the Python 2. If you have different Python 2 and/or Python 3 installation it could still be going to a different one than what you intended.
Try to do a `pip -V` it should show you the pip version and executable location, so it'd be easy to figure out which Python directory it is using. Then if you just run the python interpreter on your command line it should print its location at the top.

catchpolej, 2016-09-26:
You are right pip -V gives me python2.7 as its version.
How do I get it to add microfs to the python 3.4 version?

carlosperate, 2016-09-26:
It will most likely be `pip3.4 install ...`, but if you'd like to see how many pip commands are availble you can type in the terminal just `pip` and then press the tab key twice, it should show you all available pip commands.

carlosperate, 2016-09-26:
It will most likely be `pip3.4`, as you have tried. You can also `pip3.4 -V` to see its location. And if you'd like to see how many pip commands are available you can type in the terminal just `pip` and then press the tab key twice, it should show you all available pip commands.

catchpolej, 2016-09-26:
I am going round in circles again now......... I might have time to try again tomorrow evening.

ukBaz, 2016-09-26:
If you use pip3 does that install it in the correct place?

catchpolej, 2016-09-27:
pip3 install microfs says requirement is already up to date but pip -V still gives 2.7 as its version.

catchpolej, 2016-09-27:
I have done a few pip updates and finally pip -V comes back with python 3.4. No joy from the microbit though and python 3.4 still does not recognise the os module.

ntoll, 2016-09-27:
Can you please paste ALL the output that you see.

catchpolej, 2016-09-27:
For example:
root@jca-HP-255-G1-Notebook-PC:/home/jca# pip3.4 install --upgrade microfs
Requirement already up-to-date: microfs in /usr/local/lib/python3.4/site-packages
Requirement already up-to-date: pyserial in /usr/local/lib/python3.4/site-packages (from microfs)
root@jca-HP-255-G1-Notebook-PC:/home/jca# pip3.4 -V
pip 8.1.2 from /usr/local/lib/python3.4/site-packages (python 3.4)
root@jca-HP-255-G1-Notebook-PC:/home/jca# python
Python 3.4.4 (default, Sep 22 2016, 15:18:56) 
[GCC 4.6.3] on linux
Type "help", "copyright", "credits" or "license" for more information.
>>> ufs ls
  File "<stdin>", line 1
    ufs ls
         ^
SyntaxError: invalid syntax
>>> ufs
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
NameError: name 'ufs' is not defined
>>>

carlosperate, 2016-09-27:
`ufs ls` and similar commands are meant to be run directly from the command line, not within python. If you can `import microfs` in the python interpreter successfully it means it's installed.

ntoll, 2016-09-27:
Viz https://microfs.readthedocs.io/en/latest/#usage

```
from microfs import ls, rm, put, get
```

You're using the Python shell in the example you pasted. To use the `ufs` command just use if from your regular shell, presumably like this:

```
root@jca-HP-255-G1-Notebook-PC:/home/jca# ufs ls
```

catchpolej, 2016-09-27:
Aaaahh! Thank you for your patience and perseverance. I have managed now to use Mu to flash a python program onto the microbit that creates a file and use ufs to get the file. This is what we need for our project. Thanks again.

closed 2016-09-27 by catchpolej

ntoll, 2016-09-28:
@catchpolej nice one!

It shouldn't be that hard. Any feedback you have on how to make this process easier would be most appreciated. :-)

catchpolej, 2016-10-06:
We have started using the microbits in STEM classes now. Going fine. In terms of feedback I was wondering if I got confused partly because in http://microfs.readthedocs.io/en/latest/ at the start of the Usage section the "In your code" part and "from the command line" part are not visually separated.