bbcmicrobit/micropython #349: Missing input() builtin?

zejn, 2016-10-01:
Hi,

I'm trying to use input() from user when connecting over serial, but it seems it's not implemented in bbcmicrobit port. Am I reading the code correctly?

ZanderBrown, 2016-10-01:
Yes it isn't implemented. Look at the UART module, Personally haven't had much luck with it though

ntoll, 2016-10-01:
Docs are here:

https://microbit-micropython.readthedocs.io/en/latest/uart.html

What exactly are you trying to achieve..?

ZanderBrown, 2016-10-01:
What @zejn is trying to do i don't know but if you have a working `uart.readline()` sample that would be great. So far i have been only been able to get a single character from it.

zejn, 2016-10-02:
Ah, I see now. Instead of using input(), one can use uart.read() to read input over serial. Works great for me.

With this I was able to create a simple serial chat protocol where I can instruct one microbit to send something over the radio and make the other receive and display it.

Thanks. :) :+1:

closed 2016-10-02 by zejn

ZanderBrown, 2016-10-02:
Just what I've been doing! 😄 
[image] (The box is supposed to be a bell (`␇`) but is displaying as a box)

Complete with Animations for message received, and whether or not the mb just sent or received.