bbcmicrobit/micropython #350: Background Processes

damianburrin, 2016-10-02:
Is it possible to run a process in the background as per the blcoks (In background) command.?

Thanks
Damian

ntoll, 2016-10-02:
Some methods that take a long time and/or block the process have a `wait` flag. If set to `False` these methods run in the background. e.g. `display.scroll`, `display.show` or `music.play` all do this.

Hope this helps. Is there something special you had in mind..?

damianburrin, 2016-10-02:
I think that may solve my problem.

It’s almost Christmas so – NEOPIX LED and peizo buzzer to play jingle bells at the same time.  Will soon irritate the parents!!

Thanks
Damian

From: Nicholas Tollervey 
Sent: Sunday, October 2, 2016 4:56 PM
To: bbcmicrobit/micropython 
Cc: Damian ; Author 
Subject: Re: [bbcmicrobit/micropython] Background Processes (#350)

Some methods that take a long time and/or block the process have a wait flag. If set to False these methods run in the background. e.g. display.scroll, display.show or music.play all do this.

Hope this helps. Is there something special you had in mind..?

damianburrin, 2016-10-02:
Ah – not quite working as i’d hoped.  In PXT i can write a forever loop for the sound and play it in the background and then have a forever loop doing the display in the foreground.  I was trying to do the same in Python but it doesn’t seem possibe as its doesn’t seem the wait is actually a background thread just don’t wait unit its finsished to carry on.

I will have to think of another way

Thanks
Damian

From: Nicholas Tollervey 
Sent: Sunday, October 2, 2016 4:56 PM
To: bbcmicrobit/micropython 
Cc: Damian ; Author 
Subject: Re: [bbcmicrobit/micropython] Background Processes (#350)

Some methods that take a long time and/or block the process have a wait flag. If set to False these methods run in the background. e.g. display.scroll, display.show or music.play all do this.

Hope this helps. Is there something special you had in mind..?

dpgeorge, 2016-10-03:
Threads are not supported in MicroPython on the microbit.

But you can do `wait=False` and `loop=True` for some methods to run them forever in the background.

jtyr, 2016-10-05:
Something like callback functions should be supported by MicroPython because Microsoft Block Editor supports it. The implementation in Python should allow to do something like this:

```python
from microbit import *

# Global variable holding the index
INDEX = 0

def dec_index():
    """Callback function to decreases the INDEX variable"""
    if INDEX == 0:
        INDEX = 11
    else:
        INDEX -= 1

def inc_index():
    """Callback function to increases the INDEX variable"""
    if INDEX == 11:
        INDEX = 0
    else:
        INDEX += 1

def main():
    # Set actions performed when the button is pressed
    button_a.on_pressed(dec_index)
    button_b.on_pressed(inc_index)

    # List of images to show depending on the INDEX variable
    hours = (
        Image.CLOCK1,
        Image.CLOCK2,
        Image.CLOCK3,
        Image.CLOCK4,
        Image.CLOCK5,
        Image.CLOCK6,
        Image.CLOCK7,
        Image.CLOCK8,
        Image.CLOCK9,
        Image.CLOCK10,
        Image.CLOCK11,
        Image.CLOCK12
    )

    # Infinite loop to show the images
    while True:
        display.show(hours[INDEX])
        sleep(1000)

if __name__ == '__main__':
    main()
```

dpgeorge, 2016-10-05:
> Something like callback functions should be supported by MicroPython because Microsoft Block Editor supports it.

I don't see how that logic follows... Python has a different execution model to TD/JavaScript and callbacks just make things unnecessarily complex.  The microbit was made to be easy for children to understand.

In your example it would be much easier to read and write if the `while True` loop just checked the state of the button and increased/decrease the variable appropriately.

jtyr, 2016-10-05:
I know that in that particular example it would be possible to do everything in the same loop. I was just trying to show that asynchronous functions for the buttons could open completely new possibilities.

markshannon, 2016-10-05:
Both asynchronous and synchronous models are Turing complete, so the possibilities of both are limitless :smile:

Here is the synchronous version of your example.
```
from microbit import *

# Global variable holding the index
index = 0

def main():
    # Infinite loop to show the images
    while True:
        index = (index + button_b.get_presses() - button_a.get_presses()) % 12
        display.show(Image.ALL_CLOCKS[index])
        sleep(1000)

if __name__ == '__main__':
    main()
```
Whether it is "better" is subjective, but we do think that it is easier to teach.

jtyr, 2016-10-05:
I think it would be much cleaner to do it through the asynchronous calls. Btw, you have to put the `index = 0` into the `main()` function or declare the `index` variable as `global` in the `main()` function ;o)

whaleygeek, 2016-10-06:
Hi,

We did look at all the various concurrency models when designing the
micro:bit offering for MicroPython. I remember there being quite a lot of
discussion about the merits of every different possible method, and the one
we implemented won out of simplicity, as the coding interface is designed
for new learners, often children.

It's possible that you may be able to find these discussions and follow
through our reasoning and decisions at the time, as I think they are
captured in the history. They are either in the mailing list archives or in
some of the older github tickets.

Hope this helps!

David

ntoll, 2016-10-06:
Hi,

@jtyr, my 2c and many thanks for getting involved.

As @whaleygeek mentioned we spent a lot of time discussing concurrency models. As @dpgeorge correctly states (and with whom I am in 100% agreement) our aim was to make the micro:bit easy for children to understand. While *you* may have opinions about the best concurrency model as a developer, this may not be the best model for children given that we want kids to understand what's going on. 

That's why:

* The "event loop" has to be explicitly defined so kids can see *how* the micro:bit is executing the code and *why* they might need such a thing.
* Encouraging synchronous code as we do makes it much easier to reason about things if you are a beginner programmer and/or a teacher who needs to explain this sort of thing.
* More advanced learners are given an opportunity to run up against problems with such a synchronous outlook and will therefore *understand* the reasoning behind event driven concurrency models and other potential solutions (i.e. we're encouraging exploratory discovery rather than a didactic "this is the best way" approach).
* Python has excellent support for asynchronous style programming (co-routines, asyncio, Twisted etc...) and the skills they learn on the micro:bit can easily be re-used with more extensive implementations of Python (including MicroPython on other boards and, of course, CPython).

Hope this makes it clear why we're doing things the way we are with MicroPython on the micro:bit. Of course, we all want "many flowers to bloom" which is why there are alternatives and choices in terms of platforms people can use.

Best wishes!

N.

jtyr, 2016-10-06:
Thanks for taking the time to share all the reason with me. I fully understand why you went for the simple way but I also found a bit pity that things which are possible in the Microsoft Block Editor and Microsoft Touch Develop are not possible in python. It would be really nice if one could write the same program in all the editors.

ZanderBrown, 2016-10-06:
I at first thought this was a limitation but I've actually found it to be better as it's clearer what code runs when and it's not that hard to create your own miniature event loop

damianburrin, 2016-10-06:
Thanks for the interesting discussion chaps.  In the end thats to help from
@ntoll we were able to come up with a solution that replicates the PXT method
and is straightforward for the kids to do.

#Demonstration of playing sound, dispalying text and

#neopixel LED as three processes running together

#D Burrin 3/10/2016

from microbit import *

import music

import neopixel

from random import randint

music.set_tempo(ticks=4, bpm=120)

# play Jingle Bells - (c) James Peirpont 1857.

jingle = [ 'e:2','e:2','e:4', 'e:2','e:2','e:4', 'e:2','g:2','c','d', 'e:8',
'f:2','f:2','f:3', 'f:1','f:2','e:2', 'e:2','e:1','e:1', 'e:2','d:2','d:2',
'e:2','d:4','g:4', 'e:2','e:2','e:4', 'e:2','e:2','e:4', 'e:2','g:2','c','d',
'e:8','f:2','f:2','f:2', 'f:2','f:2','e:2', 'e:2','e:1','e:1',
'g:2','g:2','f:2', 'd:2', 'c:4' ]

#wait set to not stop for song to finish, loop set to True so repeats

music.play(jingle,wait=False,loop=True) 

display.scroll(“merry christmas”,wait=False,loop=True)

np = neopixel.NeoPixel(pin1, 8)

while True:

      for pixel_id in range(0, len(np)): 

            red = randint(0, 255)

           green = randint(0, 255)

           blue = randint(0, 255)

          # Assign the current LED a random red, green and blue value between 0
and 60

          np[pixel_id] = (red, green, blue)

           # Display the current pixel data on the Neopixel strip

           np.show()

          sleep(5)

markshannon, 2016-10-11:
@damianburrin if you're happy with the status quo, could you close this issue please.

damianburrin, 2016-10-11:
Thanks chaps - I think all is good here

closed 2016-10-11 by damianburrin