bbcmicrobit/micropython #351: speech.sing with an invalid phoneme, prevents future singing

whaleygeek, 2016-10-02:
[image]

If you use speech.sing() with an invalid phoneme, future speech.sing commands return a ValueError: Phoneme not understood.

MicroPython v1.7.9-gbe020eb on 2016-04018;

[image]

As taken from codewith.mu MacOsX version downloaded on Sun 2nd Oct 16:30

Reproducible.

ntoll, 2016-10-02:
Yeah... I've just seen this too.

For me, I do:

```
>>> import speech
>>> speech.sing("Hello")
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
ValueError: Phoneme not understood
>>> speech.sing("#115DOWWW")
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
ValueError: Phoneme not understood
```

At which point the speech synthesiser sings the correct phoneme but plays a continuous tone at a pitch somewhere between a Bb and B natural (as far as I can tell). 

If I continue to enter commands into the REPL, the speech synthesiser throws the error, correctly plays the expected sounds before reverting to the continuous tone. E.g.

```
>>> speech.say("Hello")
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
ValueError: Phoneme not understood
```

...results in "Hello" being said then the tone. Interestingly, the following worked but dropped the continuous by exactly an octave.

```
>>> speech.say("Hello there how are you")
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
ValueError: Phoneme not understood
```

Finally, the following...

```
>>> speech.say("Hello there how are you, my name is micro:bit")
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
ValueError: Phoneme not understood
```

...is a long fragment of speech that *does not* end with a continuous tone. However, any further shortish speech related interaction reverts back to the continuous tone.

Interesting. @markshannon, any thoughts on this? Obviously, some sort of error state isn't getting cleared.

whaleygeek, 2016-10-02:
That error message is generated at this line in sam.c

https://github.com/bbcmicrobit/micropython/blob/07ff6cc67fe4c0091ef6708d2b94ea281dffb46f/source/lib/sam/sam.c#L498

ntoll, 2016-10-03:
@whaleygeek see #352 for @markshannon's fix. We're launching the Christmas website at work today (! - the thing I've been working on since January). Ergo today = busy, although I should be able to test it / merge over lunch or this evening.

closed 2016-10-03 by ntoll

ntoll, 2016-10-03:
Resolved since #352 is merged into master. Will update downstream apps (uflash / Mu) this evening when I return from work.

commit c726b746 (2016-10-03): Clean speech exception state at start and end. Fixes #351.