bbcmicrobit/micropython #354: PWM Help

damianburrin, 2016-10-09:
OK bare with me here please chaps as i'm on the edge of my understanding and am trying to learn :-)

I've read this https://microbit-micropython.readthedocs.io/en/latest/pin.html and been looking at the Mbed C IO section https://lancaster-university.github.io/microbit-docs/ubit/io/

I'm trying to build a camera gimbal and have a really rough prototype running but would like to fine tune it a lot more.  At the moments i'm getting the X axis and then trying to convert it accross to a value that wil give an appropraite counter adjustment and write it back to the analog pin0.  This is ok but it's a vaule between 0 and 1023 for 0% - 100% on  PWM.  The servo I'm currently playing with has a 180d rotation with a PWM period of 20 and a range of 550uSec to 2250uSec with 1500uSec being in the middle.  Looking at the DAL IO There seems to be a functoib setServoValue that alows me to use a value between 0 and 180 (500 - 2500uSec) directly. Are these functions available via Python or am i mussunderstanding somthing along the way.

Any help appreciated.

Thanks
Damian

deshipu, 2016-10-09:
There are no such functions, but they are trivial to write -- it's just a simple proportion.

deshipu, 2016-10-09:
You can take a look at my Servo class for the Micro:bot here: https://bitbucket.org/thesheep/micro-bot/src/6bd959f7753019654f0e2e2e449211282b0496c5/bot.py?at=default&fileviewer=file-view-default#bot.py-4:13

markshannon, 2016-10-09:
They are trivial to write, but we might as well add them. We should be able to achieve slightly better precision than `analog_write()` can provide.

markshannon, 2016-10-09:
FTR the DAL provides three variants of `setServoValue()` as well as `setServoPulseUs()`, I intend to provide just `set_servo_pulse_us()`. We can document how to do the trivial maths required.

damianburrin, 2016-10-09:
I will have a play, like i said willing to learn.

deshipu, 2016-10-09:
The problem is, the exact values for minimum and maximum pulse width, as well as how many microseconds correspond to how many degrees is not standard -- every servo model has them different. I've even seen differences between the same servo model from different manufacturers. On ESP8266 I work around that by letting the user specify the minimum pulse width, maximum pulse width and the total range that corresponds to in degrees (see https://bitbucket.org/thesheep/pony/src/1529d6a904482d8a03089c472e6576efd347431b/servo.py?at=default&fileviewer=file-view-default).

deshipu, 2016-10-09:
Ah, just `set_servo_pulse_us` sounds good!

deshipu, 2016-10-09:
Do we also want to support different frequencies than 50Hz? If I remember correctly, digital servos can do up to 300Hz, which results in much smoother movement.

Darshnik253, 2022-02-02:
i am making robotic car project in that i have to control dc motor with pwm signal using microbit. so i have written code for that which i combined digital write and analog write pins which part i am attaching below. But code is working only one time after that its not working(error - microbit led is flickering). how can i resolve this error?

pins.digital_write_pin(DigitalPin.P1,0)
pins.digital_write_pin(DigitalPin.P2,0)
pins.digital_write_pin(DigitalPin.P8,1)
pins.digital_write_pin(DigitalPin.P12,0)
pins.analog_write_pin(AnalogPin.P0,511)
pins.analog_set_period(AnalogPin.P0,200)
pins.analog_write_pin(AnalogPin.P3,511)
pins.analog_set_period(AnalogPin.P3,200)