bbcmicrobit/micropython #357: Using radio.send/receive in a tight loop results in "data packet received is not a string"

deshipu, 2016-10-11:
See http://forum.micropython.org/viewtopic.php?f=17&t=2507&p=14691 for the original report.

The minimal code to reproduce is as follows. On the first microbit:

```
>>> import radio
>>> radio.on()
>>> while True: radio.send("abc")
```

On the second microbit:

```
>>> import radio
>>> radio.on()
>>> while True: radio.receive()      
... 
'abc'
'abc'
'abc'
'abc'
'abc'
'abc'
'abc'
'abc'
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
ValueError: received packet is not a string
```

The problem doesn't materialize when using `radio.receive_bytes()` or when there is a delay after every send.

deshipu, 2016-10-11:
Actually, the problem is visible when using `radio.receive_bytes()`, I just didn't see it immediately:

```
>>> for i in range(16): radio.receive_bytes()
... 
b'\x01\x00\x01abc'
b'\x01\x00\x01abc'
b'\x01\x00\x01abc'
b'\x01\x00\x01abc'
b'\x01\x00\x01abc'
b'\x01\x00\x01abc'
b'\x01\x00\x01abc'
b'\x01\x00\x01abc'
b'\x01\x00\x01abc'
b'\x01\x00\x01abc'
b'\x01\x00\x01abc'
b'\x01\x00\x01abc'
b'\x01\x00\x01abc'
b'a'
b'\x01\x00\x01abc'
b'\x01\x00\x01abc'
>>> for i in range(16): radio.receive_bytes()
... 
b'\x01\x00\x01abc'
b'\x01\x00\x01abc'
b'\x01\x00\x01abc'
b'\x01\x00\x01abc'
b'\x01\x00\x01abc'
b'\x01\x00\x01abc'
b'\x01\x00\x01abc'
b'\x01\x00\x01abc'
b'\x01\x00\x01abc'
b'\x01\x00\x01abc'
b'\x01\x00\x01abc'
b'\x01\x00\x01abc'
b'\x01\x00\x01abc'
b''
b'\x01\x00\x01abc'
b'\x01\x00\x01abc'
```

markshannon, 2016-11-07:
I cannot reproduce this on master. 
@deshipu What build are you using?

deshipu, 2016-11-07:
I was using whatever was the default for Mu at the time. Looking at one of the boards that still has the code, I can see:

```
MicroPython v1.7-9-gbe020eb on 2016-04-18; micro:bit with nRF51822
```

carlosperate, 2016-11-07:
At the moment that doesn't give much info about the microbit micropython build (more info on https://github.com/bbcmicrobit/micropython/issues/370). You'll probably have better luck tracking the version of Mu you used and looking at the dates to deduce micropython commit.

markshannon, 2016-11-07:
@deshipu Can you reproduce with a master build?

deshipu, 2016-11-07:
I will try.

deshipu, 2016-11-07:
I couldn't reproduce this with the current Master of Mu. I guess it fixed itself, closing.

closed 2016-11-07 by deshipu