bbcmicrobit/micropython #359: how representation 'a' in micro:bit use micro:python?

KimHwaMok, 2016-10-13:
how representation 'a' in micro:bit use micro:python?
i looking at micro:python library, but little bit complicated
Lastly my goal is new micro:python module!

dpgeorge, 2016-10-13:
> how representation 'a' in micro:bit use micro:python?

I don't really understand this question, but if you're refering to the object representation (MICROPY_OBJ_REPR) then it's possible to change it by editting the inc/microbit/mpconfigport.h file.

ntoll, 2016-10-13:
Hello @KimHwaMok,

Do you mean a new Python module or C module?

If you mean a Python module you can upload the Python file to the microbit using Mu (http://codewith.mu/) or microfs (https://microfs.readthedocs.io/en/latest/). Once the file is on the micro:bit's "file system" you can import it like in normal Python.

An example is given here: https://microbit-micropython.readthedocs.io/en/latest/tutorials/storage.html#mainly-main-py

Hope this help! Happy to answer more questions! :-)

closed 2016-10-14 by KimHwaMok

ZanderBrown, 2016-10-14:
is #360 related?

KimHwaMok, 2016-10-19:
yes... but new one is more detailed.