bbcmicrobit/micropython #361: No RSSI measurement on the radio module?

whaleygeek, 2016-10-20:
I'd like to do received signal strength indication measurements from the radio module on MicroPython on the micro:bit.

There doesn't seem to be anything in the API to retrieve this information, the most I can find from a github search is this single reference:

https://github.com/bbcmicrobit/micropython/blob/a20fbe82c919cea9519ce5a1140103f518b51caa/source/microbit/modradio.cpp#L78

Would it be possible for someone to add a feature to get the RSSI level of the last received radio packet? It seems like the structure record in the above line of code would surface this reading, providing it was then scaled to the same range that PXT offers.

I and others would like to use RSSI measurements with 3 micro:bits to explain how triangulation works in cell towers.

ukBaz, 2016-10-20:
Sorry, a bit of an aside. Is it triangulation or trilateration you are looking to do with RSSI?
http://circuitcellar.com/ee-tips/triangulation-trilateration-or-multilateration-ee-tip-125/

dpgeorge, 2016-10-21:
Yes, there is the capability for this, and I actually tested it when developing the radio module, but it never got exposed on the Python side.

Can someone please propose a method name and behaviour for this?

ukBaz, 2016-10-21:
My suggestion would be that the library present 'distance' to users rather than raw RSSI.
From my experience of doing a few Bluetooth beacon workshops where we worked on proximity, the students never got beyond just using the distance value from the library.

For my workshop I used:
distance = math.pow(10, (rssi - tx_power_at_1m) / (-10 * path_loss_exponent))

Which was based on these articles:
http://matts-soup.blogspot.co.uk/2013/12/finding-distance-from-rssi.html
And section two of:
http://www.intechopen.com/books/emerging-communications-for-wireless-sensor-networks/indoor-location-tracking-using-received-signal-strength-indicator

Just to set peoples expectation, there will be a lot of fluctuation on RSSI readings especially in an indoor environment with students moving around.

My thought on implementation is that it is a characteristic of the receive() function. Maybe something like this:
```
incoming = radio.receive()
if incoming['message'] == 'flash' and incoming['distance'] < 2:
```

However I appreciate that this would break backwards compatibility so maybe you have have to set parameter in `rado.receive(message_only=False)` to get the additional information. 
This would allow the library to expose RSSI and distance.

whaleygeek, 2016-10-21:
Thanks @dpgeorge 

I forget exactly what the design rules are for choosing method names, and method names over properties etc, so please do 'filter' my suggestions as appropriate. However, as always, there are some decisions to be made. So I'll try and map out the landscape and get comments from others, so that we can work this up into a proposal together.

Technically, it is RSSI, but I don't think that means anything to the target age group, even though it is technically correct.

The PXT editor calls it 'received signal strength', which is child friendly and correct enough. However, for a Python method name, that could be quite a lot of typing (although I note there is now code completion both in the REPL and in Mu so that might not be quite as much of a problem).

What is important to note, is that this is actually the RSSI of the *last received packet*. So really it is either a property of the packet (but that gets hard as the packet is just bytes), or it is stored state within the radio module. The latter is more in line with how PXT surfaces this functionality. 

To be consistent with other modules, it should be called get_ something.

The closest existing API method I can find is compass.get_field_strength().

We could call it radio.get_signal_strength(), that is consistent. However it is a *bit* misleading, as it is not the live signal strength of the radio channel, it is the signal strength of the last received packet, which could have been several seconds or even minutes away, and the value will only be updated with another call to receive().

From a simplicity point of view, and a correctness point of view, it could be:

radio.get_last_signal_strength()   (as I think that get_last_received_signal_strength is far too long!).

This raises the question, should we bundle a timestamp with it, so we know how long ago the sample was taken (e.g. you could compare it against microbit.running_time() to see how long ago the received signal strength was sampled, and make an assessment as to whether it is a good enough estimate to base decisions on or not. I'm not sure I can find any Python API methods that return bundled results, to know how to suggest that.

With a proviso that perhaps we could also add in the future radio.get_last_receive_time() which returns the microbit.running_time() timestamp of when the last packet was actually received.

Now, I note from the radio API that there is a message queue. So strictly speaking, the radio module would need to store the RSSI in the message queue alongside the message payload (and perhaps the timestamp that it was received). radio.receive() would peel off the next packet from the message queue, load the timestamp and the RSSI into module state, and then return the packet. If there was nothing in the receive queue at the point of receive, presumably this would set the timestamp and RSSI state to None? (Although comparing with None causes some other challenges, but presumably we want our audience to deal with those challenges rather than sidestepping them?)

So, as a first proposal, how about:

radio.receive()
radio.get_last_signal_strength() -> rssi value (see below)
radio.get_last_receive_time() -> time compatible with microbit.running_time()

For RSSI values, I think it would make sense to be consistent with the values returned by PXT and/or the DAL, so that it is easy to port examples between different languages.

PXT returns the signal strength as a number from 0 to 255, but note that the range is such that 255 is the weakest, and 0 is the strongest. I'm open to discussion on this. The range seems counter-intuitive to a young person. But consistency with other editors might be desirable. Although if one of the design goals of MP on micro:bit is 'no surprises', and that sometimes overrides 'strict correctness' or 'inter-editor-compatibility', that could make a case for using 0 to mean no signal, 255 to mean the strongest signal, and None to mean there is no sample. (There is additionally complexity in how PXT interprets the RSSI value, see further below)

RSSI, by industry convention, is represented as 0 for the strongest signal (the reference signal), and as a negative number for values below that. 

https://en.wikipedia.org/wiki/Received_signal_strength_indication

The value returned by PXT is not completely in following with this convention (I think?), as in two's compliment, the number 255 represents -1 and 254 represents -2 (meaning that 255 is actually a bigger number than 254). whereas in PXT, 255 represents the weakest signal, and 254 represents a slightly stronger signal!

This would break compatibility with other languages. I note that in the DAL for example, the docs don't state how to interpret the RSSI value: https://lancaster-university.github.io/microbit-docs/ubit/radio/#getrssi

Looking at the DAL code, it gets the RSSI value directly from the NRF radio structure here: https://github.com/lancaster-university/microbit-dal/blob/master/source/drivers/MicroBitRadio.cpp#L75

If the CRC of the payload fails, it (confusingly?) sets the RSSI to zero (which implies maximum signal strength??) here: https://github.com/lancaster-university/microbit-dal/blob/master/source/drivers/MicroBitRadio.cpp#L90

I've just logged an issue with the DAL to add an explanation of the range of getRSSI to the docs, to clarify this. Here is the issue number:

https://github.com/lancaster-university/microbit-dal/issues/226

If anyone has any comments on this, please chip in. I'd like to think this was more of a consensus than a 1 man proposal, as I note from other previous MicroPython API discussions, that this has always lead to a really good API decision being made.

ntoll, 2016-10-21:
Folks, I'm going to try to take what you've suggested and turn it into a pull request to our documentation so we have a place to discuss the actual API design. I'll reference the PR here when I've created it.

deshipu, 2016-10-21:
Another option would be to have an "advanced" `receive` function, either with a flag or a separate method, that would return a tuple of the timestamp, message and signal strength. I don't have any idea about how to call it, though, maybe something like "receive_details"?

If the value of the strength is reversed, maybe we could call it accordingly? "Weakness" would be the most straightforward, but maybe a little too confusing. "Signal loss" comes to mind. "Attenuation" is certainly too big a word to use here.

deshipu, 2016-10-21:
Come to think about it, I'm not sure it's worth wasting the memory storing the timestamps and strength in the queue just to have a "correct" solution, when this will actually be very rarely used. Most of the time all we are interested in is the strength of the connection at the moment, and we don't really care for which message exactly it was... Unless we want to triangulate (actually trilaterate) the position based on signals strengths from several microbits -- then we care which one is which...

ntoll, 2016-10-21:
@deshipu I believe it is exactly this use case that @whaleygeek is thinking of. Give me 5mins and I'll have something for people to discuss in terms of API docs.

ntoll, 2016-10-21:
See #362 for the PR relating to this.

whaleygeek, 2016-10-21:
FYI, update on PXT use of this API: https://github.com/lancaster-university/microbit-dal/issues/226#issuecomment-255401239

ntoll, 2016-10-25:
Hi David,

You do know what I'm going to suggest don't you..?

Why not make the change yourself and ping us a pull request?

It's not meant to be unfriendly but is a compliment.

Alternatively, add an issue to our repos.

:-)

N.

shubhdwivedi, 2016-12-30:
how to read signal strength in dw1000 ,
because when i am reading values in given register it is not working properly and giving zero value
please solve this issue

dpgeorge, 2016-12-30:
> how to read signal strength in dw1000 ,

I think you might have the wrong repository... this here is for the microbit.

ntoll, 2017-02-10:
@DavidWhaleMEF it's not stalled - it's just what happens when code is maintained by volunteers. Sometimes a lot happens very quickly as people find some free time, at other times everyone is busy actually earning money so simply don't have the bandwidth. I believe @jaustin is trying to facilitate making it worth while for someone to spend time on this sort of thing. I presume @jaustin would have a list of prioritised tasks in a backlog.

dpgeorge, 2018-03-07:
radio.receive_full() was implemented in e6f9ab13baa86cab58cbd093c7f2adf91ce56af7

closed 2018-03-07 by dpgeorge