bbcmicrobit/micropython #363: Running out of heap

carlosperate, 2016-10-21:
Running the latest available gcc-arm-none-eabi 5.4.1 for debian-based systems (https://launchpad.net/~team-gcc-arm-embedded/+archive/ubuntu/ppa), I am basically running out of heap on a clean master:

```
/usr/bin/../lib/gcc/arm-none-eabi/5.4.1/../../../../arm-none-eabi/bin/ld: region RAM overflowed with stack
collect2: error: ld returned 1 exit status
ninja: build stopped: subcommand failed.
error: command ['ninja'] failed
```

I basically have to reduce 320 bytes from:

```c
// allocate the heap statically in the bss
static uint32_t heap[9692 / 4];
```

to:

```c
static uint32_t heap[9372 / 4];
```

Has anybody else experienced this?

dpgeorge, 2016-10-24:
I have to ask: did you do "yt up"?  I did see this problem before but I think it was just on a development branch.  Current master builds for me just fine, using gcc 6.2.0, but 5.4.1 should be that different that you need 320 extra bytes of RAM.

carlosperate, 2016-10-24:
Yes, I've basically used [this script](https://github.com/carlosperate/microbit-dev-env/blob/17c407ca12bcbc160b9e02ee9e6500170d5fa73f/build_source_code.sh#L19-L24) and done all this on a Vagrant virtual machine using Ubuntu 16.04, so this specific case can be easily replicated ([link](https://github.com/carlosperate/microbit-dev-env)).

I forgot to add that I was also using the patch submitted to https://github.com/lancaster-university/microbit-dal/pull/228, not sure if that could affect it.

dpgeorge, 2016-10-24:
> I forgot to add that I was also using the patch submitted to lancaster-university/microbit-dal#228, not sure if that could affect it.

Hmm, it might.  Can you please try without that patch?

carlosperate, 2016-10-24:
Can't compile micropython without it, so unless I compile a newer compatible version of gcc arm from source (or install arch) I am not sure how else to test it.
Alternatively you could try to compile it in 6.2 with the changed dependency?

dpgeorge, 2016-10-24:
> Alternatively you could try to compile it in 6.2 with the changed dependency?

Using "carlosperate/microbit-dal#update_ble_dep" as the dependency for microbit-dal (instead of lancaster version) it builds ok for me with gcc 6.2.

The issue you're having is probably something to do with the C library (newlib?) that's taking extra RAM.  Without inspecing the resulting .map files it's hard to debug this.

carlosperate, 2016-10-24:
Could you upload somewhere the map file you got from the build with my dal fork? I could have a look tonight and post mine as well.

dpgeorge, 2016-10-24:
> Could you upload somewhere the map file you got from the build with my dal fork?

See http://micropython.org/resources/microbit-micropython.map.gz

carlosperate, 2016-10-24:
Thanks Damien, I'll see what I can spot tonight.

carlosperate, 2016-10-25:
So I didn't have as much time as I thought tonight, so didn't have a chance to have a proper look at the map files. On a quick scan noticed that the data section is ~360 Bytes larger:

```
mine:   .data           0x0000000020000000      0x290 load address 0x000000000003c404
yours:  .data           0x0000000020000000      0x128 load address 0x000000000003a238
```
But the bss is 316 bytes smaller
```
mine:   .bss            0x0000000020000290     0x2d70 load address 0x000000000003c694
yours:  .bss            0x0000000020000128     0x2eac load address 0x000000000003a360
```

https://gist.githubusercontent.com/carlosperate/bee7267830dbc1aed8d3b828e95cfb10/raw/96ac2b88cfa939c5fe218502d7c9d2085ac0ca6b/microbit-micropython.map

Is it possible that the `std` has not been set and the default has changed to c++11 or 14 between gcc 5 and 6?

dpgeorge, 2016-10-25:
Make sure you build your version with the original uPy heap size.  To get it to build you can edit ./yotta_targets/bbc-microbit-classic-gcc-nosd/ld/NRF51822.ld and change line 6, increase the RAM LENGTH.

carlosperate, 2016-10-28:
Sorry, I haven't had a chance to look into this yet (had to spend the time working on something else instead). I'll see if I can spare some time next week.

@dpgeorge we could still target the newest version of the DAL as that should work for everybody as this is a different issue. It's a tiny change, do you you'll change that manually soon, or prefer I send a PR?

dpgeorge, 2016-11-03:
@carlosperate see #374 for the update to the DAL.

dpgeorge, 2017-01-06:
I also hit this (again) in the past few days.  I'll try to find a fix.

dpgeorge, 2017-01-06:
> Reverting to newlib 2.4.0-2 solved it for me (all the locale symbols disappeared from the map). (Note that reverting to 2.4.0-4 wasn't sufficient).

newlib 2.5.0 introduced some new locale handling code, and has a large (~350 bytes) `__global_locale` data structure that is not constant and isn't put in ROM but rather in RAM.  Looking at the newlib source code it could be that this data structure is read-only and could in fact go in ROM, but that requires patching/fixing upstream (and it's a long way upstream, would take a while to filter down).

There is no easy way to fix this.  The data structure is used by strtol, isalpha, etc and you can't override these symbols by defining your own.  You could disabled the entire stdlib (-nostdlib) but then you'd need to provide implementations of many functions, like memset, sprintf, fclose, malloc, free, etc.

Only real option seems to be to stick with newlib 2.4.0-2.

@jaustin any other ideas how to fix or work around this?

dpgeorge, 2017-01-19:
Ping @jaustin : this issue with newlib 2.5 taking a lot of RAM may soon become a real issue for everyone using mbed, not just us.  Do you have any thoughts how to deal with it?

dpgeorge, 2018-03-07:
Fixed by 498125d2619f5f7a48026f12d6c984dd476643d7

closed 2018-03-07 by dpgeorge

karlp, 2019-08-22:
@dpgeorge fwiw, upstreams has "won'tfix"'d it anyway :( https://bugs.launchpad.net/gcc-arm-embedded/+bug/1686067

dpgeorge, 2019-08-23:
Thanks @karlp for the pointer.  That is a shame it won't be fixed upstream.