bbcmicrobit/micropython #364: wrong link to help docs in help()

whaleygeek, 2016-10-23:
When I converse with MicroPython at the REPL and type

help()

I get this link to the docs:

Python/micro:bit documentation is here: http://microbit-micropython.rtfd.org/

But the official documentation is here:

https://microbit-micropython.readthedocs.io/en/latest/

I know this is just a redirector, but RTFD actually means something specific, that is probably not appropriate for the target age group:

http://www.internetslang.com/RTFD-meaning-definition.asp

I did raise this many months ago via email, but I think it has just been forgotten. So I've logged it as an issue in the hope it might be considered for correction in the next release that goes inside Mu.

Thanks

ntoll, 2016-10-23:
It's the official way in which you're supposed to shorten Read the Docs URLs that will always redirect to the latest version.

ntoll, 2016-10-23:
i.e. there's not much we can do about it. :-(

closed 2016-10-23 by ntoll

deshipu, 2016-10-23:
Actually we can use https://microbit-micropython.readthedocs.io

I just want to note, that in the case of readthedocs, the "rtfd" means "Read the Fine Documentation".

dpgeorge, 2016-10-24:
This was actually fixed about a month ago in d96be5f5af287efa7786db19f43a7a913f4394a4.

ZanderBrown, 2016-10-24:
This maybe slightly silly but

Switching from
`http://microbit-micropython.rtfd.org/` to
`https://microbit-micropython.readthedocs.io/`
Adds 7 extra bytes, something we clearly want to avoid with the memory currently available to us

deshipu, 2016-10-24:
If this is a concern, we can use an url shortener or I can deploy the documentation on a shorter url, such as "microbit-upy.readthedocs.io", for example, or even just "microbit.readthedocs.io", as readthedocs is mostly used by python projects anyways.

dpgeorge, 2016-10-26:
> Adds 7 extra bytes, something we clearly want to avoid with the memory currently available to us

It's flash memory that's used here, not RAM.  We can definitely afford 7 bytes of flash to have a proper URL :)