bbcmicrobit/micropython #367: round() function gives inconsistent (and incorrect) results sometimes

whaleygeek, 2016-10-28:
microbit.org-ticket: 178

TEST PROGRAM (using latest version of Mu as of 24 Oct 2016)

from microbit import *
# test of the rounding function to demonstrate errors
scale = 1001
for i in range(1, 100):
    result = round((scale/i),1)
    print("%s %s" % (i, str(result)))

RESULTS
1 1001.0
2 500.5
3 333.6999
4 250.3
5 200.1999
6 166.8
7 143.0
8 125.1
9 111.2
10 100.1
11 91.0
12 83.39999
13 77.0
14 71.5
15 66.69998
16 62.59999
17 58.89999
18 55.59999
19 52.7
20 50.0
21 47.7
22 45.5
23 43.5
24 41.7
25 40.0
26 38.5
27 37.09999
28 35.79999
29 34.5
30 33.39999
31 32.29999
32 31.3
33 30.29999
34 29.39999
35 28.6
36 27.79999
37 27.1
38 26.3
39 25.7
40 25.0
41 24.39999
42 23.8
43 23.3
44 22.79999
45 22.2
46 21.8
47 21.3
48 20.89999
49 20.39999
50 20.0
51 19.6
52 19.3
53 18.89999
54 18.5
55 18.2
56 17.89999
57 17.6
58 17.3
59 17.0
60 16.7
61 16.39999
62 16.1
63 15.9
64 15.6
65 15.4
66 15.2
67 14.9
68 14.7
69 14.5
70 14.3
71 14.1
72 13.9
73 13.7
74 13.5
75 13.3
76 13.2
77 13.0
78 12.8
79 12.7
80 12.5
81 12.4
82 12.2
83 12.1
84 11.9
85 11.8
86 11.6
87 11.5
88 11.4
89 11.2
90 11.1
91 11.0
92 10.9
93 10.8
94 10.6
95 10.5
96 10.4
97 10.3
98 10.2
99 10.1

markshannon, 2016-10-28:
What exactly is wrong with the rounding?

whaleygeek, 2016-10-28:
It doesn't conform to the Python spec, with regards to the number of significant digits parameter??

If I do this on Python on my Mac, I get this:

Python 3.4.3 (v3.4.3:9b73f1c3e601, Feb 23 2015, 02:52:03) 
[GCC 4.2.1 (Apple Inc. build 5666) (dot 3)] on darwin
Type "help", "copyright", "credits" or "license" for more information.
>>> round(1001/53,1)
18.9
>>> round(1001.0/53.0,1)
18.9

Do the same on the micro:bit and I get:

MicroPython v1.7-9-gbe020eb on 2016-04-18; micro:bit with nRF51822
Type "help()" for more information.
>>> round(1001/53,1)
18.89999
>>> round(1001.0/53.0,1)
18.89999

round(number[, ndigits])
Return the floating point value number rounded to ndigits digits after the decimal point. If ndigits is omitted, it defaults to zero. The result is a floating point number. Values are rounded to the closest multiple of 10 to the power minus ndigits; if two multiples are equally close, rounding is done away from 0 (so, for example, round(0.5) is 1.0 and round(-0.5) is -1.0).

Note The behavior of round() for floats can be surprising: for example, round(2.675, 2) gives 2.67 instead of the expected 2.68. This is not a bug: it’s a result of the fact that most decimal fractions can’t be represented exactly as a float. See Floating Point Arithmetic: Issues and Limitations for more information.

dpgeorge, 2016-10-28:
It is actually doing the rounding correctly.  Ie, internally the number is rounded and the rounded value is stored in the floating point object.

The "issue" is that the microbit uses 30-bit floating point numbers so the accuracy is not as good as 64-bit on the desktop.  The lack of accuracy is apparent only when you print the number using the standard "str" format.  Instead you can use "%.1f" % number to get exactly 1 decimal place in the printing.

whaleygeek, 2016-10-28:
Thanks @dpgeorge I take your point. But is that really correct and 'unsurprising' from a Python perspective? The ndigits parameter states it is the number of digits to round to, and 18.9 is a number that can be represented in the floating point internal representation, even with 30 bit representation. However, the method returns more digits than the user of the method requested in it's interface. i.e. it is a 'surprise'.

For the built-in types supporting round(), values are rounded to the closest multiple of 10 to the power minus ndigits; if two multiples are equally close, rounding is done toward the even choice (so, for example, both round(0.5) and round(-0.5) are 0, and round(1.5) is 2). The return value is an integer if called with one argument, otherwise of the same type as number.

markshannon, 2016-10-28:
Presenting the minimum length string for a floating point number is a surprisingly complex problem:
https://bugs.python.org/issue1580

whaleygeek, 2016-10-28:
Ok, sure, understood. I'll pass the work-around to the customer, thanks.

whaleygeek, 2016-10-28:
Just as an aside, this is quite interesting, given that 18.9 is indeed possible to represent as a float?

MicroPython v1.7-9-gbe020eb on 2016-04-18; micro:bit with nRF51822
Type "help()" for more information.
>>> float("18.9")
18.89999
>>> 18.9
18.89999

whaleygeek, 2016-10-28:
Sorry, md messed up the paste, try this

    MicroPython v1.7-9-gbe020eb on 2016-04-18; micro:bit with nRF51822
    Type "help()" for more information. 
    >>> float("18.9")
    18.89999
    >>> 18.9
    18.89999

whaleygeek, 2016-10-28:
The work around with the print formatting works. But I just wanted to confirm that the behaviour above with the number 18.9 coming back as 18.89999 is 'expected behaviour' - if someone could confirm that, I'll move along. Thanks.

dpgeorge, 2016-11-03:
> But I just wanted to confirm that the behaviour above with the number 18.9 coming back as 18.89999 is 'expected behaviour'

Yes and no.  The microbit uses 30-bit floating point and 18.9 is parsed correctly/accurately but it's then rounded down to 18.899999 when stored in 30 bits.  

Probably the proper fix is to modify the printf code so that it takes into account there is only 30 significant bits in the number and do rounding when printing.  But this is a fix someone else will need to contribute because it's very low priority.

Floating point numbers are always going to have inaccuracies and you should use formatting like "%.2f" if you want a certain number of digits in the output.

dpgeorge, 2017-04-21:
This was fixed upstream, see https://github.com/micropython/micropython/pull/3029

dpgeorge, 2018-03-07:
Fixed in merge of version1 branch at e26d7c89d4a96de0fa0a1dd5aec024b31fc4816e

closed 2018-03-07 by dpgeorge