bbcmicrobit/micropython #368: sending and receiving strings via uart methods

whaleygeek, 2016-10-29:
microbit.org-ticket: 197

Is there a 'pythonic' way to send and receive strings via the uart object?

e.g. is this the most normal way to encode and decode string data, as the uart methods requires a 'bytes':

msg = "ABC"
uart.write(bytes(msg, "UTF-8"))

m = uart.readall()
msg = str(m, "UTF-8")

ntoll, 2016-10-29:
I usually just do `msg.encode('utf-8')` etc...

@whaleygeek a quick heads up...

This *isn't* the best place for these questions. Please only use GitHub for raising bugs, pointing out missing docs, asking for features etc... relating to MicroPython on the micro:bit. ;-)

For questions of Python style, best ask on the microbit@python.org mailing list, in either the micropython or microbit channels on IRC (Freenode) or on the micropython room on the official Slack site created by microbit.org.

It's a great question - I just want to make sure it's getting asked in the right place. :-)

whaleygeek, 2016-10-29:
Okay, so I'll change this into a request to add a note in the docs on the uart page, explaining how to send and receive strings, as it is a common question I get asked. Thanks!

whaleygeek, 2016-10-29:
@ntoll I couldn't get your example that you gave above to work?

On my Mac:

Python 3.4.3 (v3.4.3:9b73f1c3e601, Feb 23 2015, 02:52:03) 
[GCC 4.2.1 (Apple Inc. build 5666) (dot 3)] on darwin
Type "help", "copyright", "credits" or "license" for more information.
>>> msg = 'hello'
>>> msg.encode('UTF-8')
b'hello'
>>> 

Inside the Mu REPL

>>> msg = 'hello'
>>> msg.encode('UTF-8')
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
AttributeError: 'str' object has no attribute 'encode'
>>>

dpgeorge, 2016-11-03:
@whaleygeek the way you did it at the start of this issue is correct.  Use `bytes` and `str` to convert between bytes and str objects.  The encode and decode methods are deprecated (or at least legacy) in Python and not implemented in MicroPython.

But note that you can also write the str object directly, like `uart.write('hello')`.

microbit-carlos, 2018-10-18:
@whaleygeek Do you think a note about this still needs to be added to the docs? (using `bytes` and `str`)
Any suggestions on what to clarify would be most welcomed :)

whaleygeek, 2018-10-18:
the microbit python mailing list is now closed. This is a common question I see getting asked, and @ntoll says it's a great question - I'm just trying to decide the best place we should answer it!

It's a common issue that people ask me about when using MicroPython on the micro:bit to use the UART, so it might be worth just signposting it as a common pattern in the docs.

The problem is that users will Google for the fix and then land on all sorts of forums and official Python docs that don't relate to this specific version of Python on this specific device, and that ends up being misleading for what is a common requirement.

As MicroPython [on the micro:bit] differs from core Python in this aspect, it would make sense to signpost it in the official docs.

Perhaps as a small footnote against the uart.write() method is all that is needed here.

microbit-carlos, 2018-10-26:
Okay, so looking into the options of what/how to send via UART we have the following.

Sending a string:
```
>>> l = uart.write('hello world')
hello world
```

Sending bytes:
```
>>> l = uart.write(b'hello world')
hello world
```

Converting a string (in this case with a unicode character that takes more than 1 byte in UTF-8) to bytes and sending that:
```python
from microbit import *
bytes_from_str= bytes('UTF-8 char: £\n', 'UTF-8')
print(bytes_from_str)
l = uart.write(bytes_from_str)
```
Output (serial terminal has to be configured to decode UTF-8 for the `£` character to be shown correctly):
```
b'UTF-8 char: \xc2\xa3\n'
UTF-8 char: £
```

For receiving data with `uart.read()` we get a bytes, so we can convert those into a string

```python
from microbit import *

print(b'UTF-8 char: \xc2\xa3\n')
# Without an encoding this is not what we are looking for
print(str(b'UTF-8 char: \xc2\xa3\n'))
# Adding the encoding we should see the right unicode characters
print(str(b'UTF-8 char: \xc2\xa3\n', 'UTF-8'))
```
Output:
```
b'UTF-8 char: \xc2\xa3\n'
b'UTF-8 char: \xc2\xa3\n'
UTF-8 char: £
```

We can probably condense this to a few lines, but I am not quite sure where would be the best place. the intro paragraph to the UART API docs? The `read` and `write` method descriptions?

carlosperate, 2018-10-29:
@whaleygeek I've submitted PR #588, could you check it out and let me know if that is enough info? Any comments are more than welcome.

closed 2018-11-14 by jaustin