bbcmicrobit/micropython #372: Documentation for Ubuntu 16.04 is invalid

wojla, 2016-11-01:
After `sudo apt-get update`: 

`E: Failed to fetch http://ppa.launchpad.net/pmiller-opensource/ppa/ubuntu/dists/xenial/main/binary-amd64/Packages  404  Not Found`

dpgeorge, 2016-11-02:
The repository may have been offline for a bit.  Please just try again.

carlosperate, 2016-11-02:
The https://launchpad.net/~pmiller-opensource/+archive/ubuntu/ppa page doesn't show any packages for 16.04 (xenial), which is probably why it's giving the 404. For instance the same link for 14.04 is working: http://ppa.launchpad.net/pmiller-opensource/ppa/ubuntu/dists/trusty/main/binary-amd64/Packages
Do you know which package from pmiller-opensource ppa is needed ? It doesn't seem to be required for yotta and, as far as I can find, it is only instructed in the micropython instructions.

wjgeorge, 2017-05-01:
This error still persists. Please advise

markshannon, 2017-05-01:
@wjgeorge What happen if you skip that step in the instructions. What fails?

carlosperate, 2017-09-19:
I had the same issue in 16.04, so I removed adding the `pmiller-opensource/ppa` ppa and replaced installing `gcc-arm-none-eabi` with `gcc-arm-embedded` and seems to be working fine. Basically following the steps indicated to install Yotta: http://docs.yottabuild.org/#linux-cross-compile

dpgeorge, 2021-09-06:
This should be fixed by c199d6bb22330cfeb76c48c8d295eb0b3dcfe8c5

closed 2021-09-06 by dpgeorge