bbcmicrobit/micropython #373: /source/microbit/microbitdisplay.cpp module

KimHwaMok, 2016-11-02:
`DEFAULT_SCROLL_SPEED`       150

void microbit_display_scroll(microbit_display_obj_t *self, const char* str) {
    mp_obj_t iterable = scrolling_string_image_iterable(str, strlen(str), NULL, false, false);
    microbit_display_animate(self, iterable, DEFAULT_SCROLL_SPEED, false, true);
}`

dpgeorge, 2016-11-03:
Did you deploy the new firmware to your device?

KimHwaMok, 2016-11-03:
new firmware?
Is this firmware problem?
If so that how to deploy new firmware in my device?

dpgeorge, 2016-11-03:
If you recompile the code you need to reflash the microbit with the new firmware.  "make deploy" can do that but it might need to be adjusted for your system.

KimHwaMok, 2016-11-03:
@dpgeorge 
I just deployed new firmware in my device.
but my problem is same :(

dpgeorge, 2016-11-03:
You need to upload not the DAP link firmware, but the code for the nRF51.  The hex file found here: build/bbc-microbit-classic-gcc-nosd/source/microbit-micropython.hex

KimHwaMok, 2016-11-03:
@dpgeorge 
Okay, first i found microbit-micropython.hex file.
what should i change this file?

dpgeorge, 2016-11-03:
You need to download this file to the microbit, copy it to the mass storage device.

Sorry but I can't help you further beyond these instructions.

KimHwaMok, 2016-11-03:
@dpgeorge 
I understand
thank you for reply

sudo cp build/bbc-microbit-classic-gcc-nosd/source/microbit-micropython.hex /dev/ttyACM0

KimHwaMok, 2016-11-03:
@dpgeorge 
I have last question.
I followed you step everything.

i copied microbit-micropython.hex file in my mass storage device...
but but!!
i same problem...

ntoll, 2016-11-03:
Hi @KimHwaMok,

Just checking some very basic things to make sure you have everything set up correctly.

* You use `make build` to compile your code?
* If `make build` works you'll find a `firmware.hex` file in the `/build` directory.
* Copy the `firmware.hex` file to your micro:bit - this will flash *your* modified version of MicroPython onto the device.
* Try your Python code and do a little dance when it works! :-)

Sorry if all the above is obvious to you, I just want to make sure all the basics are in place.

deshipu, 2016-11-03:
> @dpgeorge 
> I understand
> thank you for reply
> 
> sudo cp build/bbc-microbit-classic-gcc-nosd/source/microbit-micropython.hex /dev/ttyACM0

No, this is wrong. The `/dev/ttyACM0` is the serial device. You want the block storage device -- the one that gets mounted somewhere in `/media` as a disk.

KimHwaMok, 2016-11-03:
@ntoll 

make build?
what is `yotta build`???

I always change /source/microbit/microbitdisplay.cpp then use `yotta build` again

KimHwaMok, 2016-11-03:
@deshipu 

thank you
In my case /media/xxx/microbit

ntoll, 2016-11-03:
Sorry, that should be `make all` (I was mis-remembering). Or better, `make yotta`. It wraps `yotta builb` with some useful stuff. Just look in the `Makefile` and you'll see.

KimHwaMok, 2016-11-03:
@ntoll 
oh my god,

i tried to `make all, make yotta` finally i found /build/firmware.hex

ntoll, 2016-11-03:
Great stuff... now copy it onto your device and try it out! :smile:

KimHwaMok, 2016-11-03:
@ntoll 

Maintenance mode!! am i right???

closed 2017-01-02 by KimHwaMok