bbcmicrobit/micropython #375: Software I2C

deshipu, 2016-11-07:
I think that i2c is a great protocol for communicating with peripherals, and it's especially nice for the microbit, where you only have very limited number of pins easily accessible -- so the ability to connect hundreds of devices with just two pins is very good. However, the pins for the hardware i2c on the microbit are in a very awkward place, and they are very hard to use in any project that doesn't utilize the bulky edge connector (and who wants to leave their edge connector in their projects, too). So I'm wondering about some way of doing i2c on the pins that are easily accessible.

The obvious way would be to have software (bit-banging) i2c, working on any pins. The upstream micropython has an implementation of this that works reasonably well -- I wonder how much sense it would make to make this available in the microbit? It will cost some flash memory, certainly, but it will make it much easier to use things like the 8x8 LED matrix backpacks, OLED displays, servo and motor controllers and sensors.

Should I try and get this to work?

markshannon, 2016-11-07:
There is no need to use bit-banging. The hardware supports using arbitrary pins, and the mbed-hal exposes a C function to re-initialise the i2c. 
See `yotta_modules/mbed-classic/targets/hal/TARGET_NORDIC/TARGET_MCU_NRF51822/i2c_api.c`.

deshipu, 2016-11-07:
Wow, thanks, I will try with that.

deshipu, 2016-11-07:
Hmm... so I tried to add an `i2c.init` method, but I'm a little stuck. There are two errors I can't quite figure out. Here is my commit: https://github.com/deshipu/micropython-1/commit/75c8c7bc6c4d0109240b6a02f2c72b7e89054e01

The first one is about the QSTRs, and is probably something trivial that I need to do in order for the strings to be picked up by the generating script.

The second is a bit more troublesome -- seems like the i2c pointer in the MicrobitI2C class is protected, so I can't call the mbed-hal functions. Is there some trick to get around that?

Any help is greatly appreciated.

dpgeorge, 2016-11-08:
> The first one is about the QSTRs, and is probably something trivial that I need to do in order for the strings to be picked up by the generating script.

You need to add them by hand to inc/microbit/qstrdefsport.h, then run "make".

> The second is a bit more troublesome -- seems like the i2c pointer in the MicrobitI2C class is protected, so I can't call the mbed-hal functions. Is there some trick to get around that?

To get around this you'd need to create your own class derived from MicroBitI2C and add a method to change the pins.

But I think it's even more difficult: if you call i2c_init then it tries to allocate a new I2C HW peripheral, not reuse the existing one.  I don't think the mbed library supports remapping of the pins...

deshipu, 2016-11-08:
Thanks for the help.

So, maybe we could go about it differently? I'm still not sure it's worth the costs, but I think it would be nice to have an easily available i2c bus (mostly because then I can use it in my robots). So I think we have several options:

- Reimplement the i2c support to drop DAL and use the mbed functions directly. I think that we will need to do that anyways, if we want to drop DAL eventually.
- Create a second i2c object, bound to the easily accessible pins (for instance, to 1 and 2). This also has the advantage that we don't interfere with the accelerometer and compass this way.
- Provide a normal constructor and let the i2c object be created on any pins as needed. Has the same advantage as the second option, but possibly uses more memory?

markshannon, 2016-11-08:
As @dpgeorge says, it does look like `i2c_init()` cannot be called repeatedly.
However the following might work:
Call `i2c_init()` initially to grab a hardware peripheral.
To change pins, call `i2c_stop()` (if necessary), set the `sda` and `scl` fields directly and then call `i2c_reset()`or `i2c_start()`.

deshipu, 2016-11-08:
@markshannon what are the plans for dropping DAL with respect to the SPI and I2C peripherals? Would it make sense for me to work on making DAL-less versions of those?

markshannon, 2016-11-08:
The DAL I2C is just a thin wrapper for the mbed I2C that includes an workaround for a hardware issue that has been fixed. You won't gain anything by using it.
I don't think the DAL has any SPI interface, but I don't really know anything about SPI so I could be wrong.

deshipu, 2016-11-08:
Ok, I will try to get rid of it then. One more question -- I can't quite find where the `uBit` object, and its `uBit.i2c` created -- I suppose there is some kind of DAL initialization code somewhere?

deshipu, 2016-11-08:
So I tried what @markshannon suggested, and added a trick to get at the protected member of the class, and lo and behold -- it works!

[image]

PR #377 (2016-11-11): Add i2c.init to change the pins and frequency
Fixes #375
closed 2016-11-11 by dpgeorge