bbcmicrobit/micropython #378: i2c.init() stops the compass working.

markshannon, 2016-11-11:
```python
>>> compass.heading()
358
>>> i2c.init(scl=pin0,sda=pin1)
>>> compass.heading()          
25
#Wiggle µBit about
>>> compass.heading()
25
>>> compass.heading()
25
```

I think we ought to revert e88e3b2f6c579eab0eafda02242092bff51f6934 until we decide how to do this properly.

deshipu, 2016-11-11:
If you look at the documentation, it's explicitly mentioned in the warning there at lines 31 and 32. This is completely expected, just like the REPL stopping working after calling `uart.init`.

deshipu, 2016-11-11:
An alternative would be to use the other I2C peripheral and have the SPI stop working.

markshannon, 2016-11-11:
Sorry, I thought that there were 4 shared i2c/spi peripherals, not just the two.
It might be nice to be able to recycle the SPI peripheral as another I2C, but put that down as a possible enhancement.

deshipu, 2016-11-11:
I was thinking about it, but I could think of no logical/elegant API for that. One crazy idea would be to have an `i2c` attribute on `microbit.spi`, so you could use `microbit.i2c` or `microbit.spi.i2c`...

deshipu, 2016-11-11:
There are only two, there are four cases in the `i2c_api.c`, because two of them are for reusing the same peripheral, and two are for using a new one.