bbcmicrobit/micropython #379: Script is too long

VlastimilHovan, 2016-11-21:
Hi All. Anybody can shine some light on message I am getting at MU ? It said script is too long to upload. Just not sure if it means too big in size or just too many lines ?
Any idea ?

Thanks a lot

carlosperate, 2016-11-21:
Hi @VlastimilHovan,

That means the script is too large, not so much the number of lines, but the number of characters. You could try to reduce the number of comments a simple fix. Alternatively if you use the "files" functionality you could upload a "main.py" script, if you use that name for the script it will automatically execute on startup, and there is more space for scripts that way. Just remember that every time you flash the microbit it will remove all the files stored there.

For future reference issues with Mu itself, like this one, are better raised in the Mu repository: https://github.com/mu-editor/mu/

Hope that helps.

ntoll, 2016-11-22:
The size restriction is in place simply because, um, we only have 256k of flash memory and not a lot of room. ;-) Carlos's excellent answer essentially gives you the options you have available.

closed 2016-11-22 by ntoll

VlastimilHovan, 2016-11-22:
Thank you very much for response.
It helps a lot