bbcmicrobit/micropython #383: read_digital() not behaving as expected

MrsAshman, 2016-12-13:
Microbit 1:
from microbit import *

while True:
    if pin2.read_digital() == 1:
        display.show(Image.HAPPY)
    else:
        display.show(Image.SAD)

Microbit 2:
from microbit import *

while True:
    if button_a.was_pressed():
        pin1.write_digital(1)  # switch the signal on
        display.scroll("1")
    if button_b.was_pressed():
        pin1.write_digital(0)  # switch the signal on
        display.scroll("0")

I may be wrong but according to the code on the website this should make the 1st microbit change the image from sad to happy when you press button A on the 2nd Microbit but it isn't working - I have the two bicrobits connected together using copper wire from pin 1 on microbit 2 and connected to pin 2 on microbit 1.

Any ideas?

Thanks 

Estelle

dpgeorge, 2016-12-13:
You also need to connect ground-ground between the 2 microbits so they have
a common reference point for the digital signal.

MrsAshman, 2016-12-14:
Thanks for that - now working!

closed 2018-03-07 by dpgeorge